cruise-automation/webviz #104: How to visualize PointCloud2 lidar?

bryanibit, 2019-03-20:
I did not install anything but I can plot and show images via [official website](https://cruise-automation.github.io/webviz/webviz-core.html). However, I can not visualize 3d point clouds whose format is pointcloud2. Could you please tell me how to achieve it? Thank you!

janpaul123, 2019-03-20:
We currently do not support visualizing 3d point clouds in the webviz-core tool, but we're working on it. Stay tuned!

calvinwyoung, 2019-05-15:
I have to visualize a terrain surface using a point cloud. I'm currently able to do this by rendering a large set of Points (downsampled of course). This works reasonably well but it's not awesome.

For this specific use case, does Worldview support rendering a mesh instead of a collection of points? It'd be cool if I could use something like the `<Grid />` component to render a mesh of the terrain.

jtbandes, 2019-05-15:
That sounds like it might be a separate issue from this one. What's the original format of your terrain data? The [Marker](http://wiki.ros.org/rviz/DisplayTypes/Marker) message has a MESH_RESOURCE type, but it's unsuitable for browser-based visualization because it points to a file on your local disk. Conceivably, support could be added to webviz for a wireframe mesh if all the data were contained in a marker message, but I'm not aware if there's a standard ROS msg type that's used for that kind of data?

calvinwyoung, 2019-05-15:
Got it, all of that that makes sense. Fwiw, *are* using the `Marker` message type -- just wanted to see if your team had a recommendation for how to convert it or pre-process it for web visualization.

jtbandes, 2019-05-16:
Which `type` of marker are you using? Keep an eye on #137 — we're close to adding our 3D panel which supports Markers out of the box (except MESH_RESOURCE).

calvinwyoung, 2019-05-16:
Ah cool, thanks for the heads up — will keep an eye on it. We're using `POINTS` internally at the moment.

janpaul123, 2019-05-21:
Should be solved per our latest update (#141). 🎉 (Feel free to file a new bug when you have a specific issue with the new 3D panel.)

closed 2019-05-21 by janpaul123

calvinwyoung, 2019-05-21:
This is awesome, thanks guys!