cruise-automation/webviz #118: How to visualize data from a live server?

baker-Xie, 2019-04-02:
We want to run our ros application, and transfer the data to worldview to visualize. But how to connect the C++ Ros node to worldview? Thanks

janpaul123, 2019-04-02:
This is currently not possible. We are working on bringing Worldview into [Webviz](https://cruise-automation.github.io/webviz/webviz-core.html) (see #104), which will allow you to drag in a recorded ROS bag file and visualize it in 3D. But connecting to a live ROS application is a bit more difficult. We might support that in the future.

Zanbrachrissik, 2019-07-20:
> This is currently not possible. We are working on bringing Worldview into [Webviz](https://cruise-automation.github.io/webviz/webviz-core.html) (see #104), which will allow you to drag in a recorded [...]

Hi, I wonder if you connected to webviz for real time visualising. I am currently working on it and I really appreciate it if you could give me some help.

jtbandes, 2019-08-04:
@Zanbrachrissik there's more discussion on this in #170.

closed 2019-12-26 by janpaul123

janpaul123, 2019-12-27:
@baker-Xie @Zanbrachrissik This is now working, see https://github.com/cruise-automation/webviz/pull/299 for details!

PR #299 (2019-12-26): Update packages/webviz-core from internal repo
This adds support for connecting to a rosbridge_server over a WebSocket connection. By default it
connects to ws://localhost:9090, but you can change this by using the `?rosbridge-websocket-url=…`
query parameter. In the future we will add better UI and documentation for this.

This fixes #118 #170 #209.

Other fixes:
- Fixed not being able to select stacked objects in the 3D panel.
- Various improvements to the rosout panel.

🎄 Happy holidays! 🎅
commit e42aa832 (2019-12-26): Update packages/webviz-core from internal repo (#299)

This adds support for connecting to a rosbridge_server over a WebSocket connection. By default it
connects to ws://localhost:9090, but you can change this by using the `?rosbridge-websocket-url=…`
query parameter. In the future we will add better UI and documentation for this.

This fixes #118 #170 #209.

Other fixes:
- Fixed not being able to select stacked objects in the 3D panel.
- Various improvements to the rosout panel.