cruise-automation/webviz #138: How to visualize ROS OccupancyGrid?

aargondizza, 2019-05-18:
Is there a Worldview command that corresponds best to ROS's OccupancyGrid? I'm thinking you could use FilledPolygons, constructing the x, y, z from the OccupancyGrid cell's indices and the width, height, position of the OccupancyGrid - then setting the alpha of the polygon from the cell value. However, it seems like the Triangles type has some nice features where it assumes triplets of points make a triangle, but I'm not sure if that works with the OccupancyGrid. Is there a type that's intended to go with OccupancyGrid?

janpaul123, 2019-05-21:
It's not part of Worldview, but in Webviz we've just [shipped](https://github.com/cruise-automation/webviz/pull/141) the 3d panel which includes an [OccupancyGrids renderer](https://github.com/cruise-automation/webviz/blob/187c3695d/packages/webviz-core/src/panels/ThreeDimensionalViz/commands/OccupancyGrids.js). Hope that solves your issue?

vidaaudrey, 2019-08-06:
Closing this issue as OccupancyGrids is now supported: https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/panels/ThreeDimensionalViz/commands/OccupancyGrids.js

closed 2019-08-06 by vidaaudrey