cruise-automation/webviz #161: No explanation of how to run

Hodapp87, 2019-06-20:
Early on it says, "Webviz is a web-based application for playback and visualization of ROS bag files." and links you to a Webviz demo, however, I cannot find any explanation of how to actually serve this web application myself and get to it from my browser.

After a bit of messing around it looks like the proper way is running `npm run docs` and then browsing to http://localhost:8080/try (just http://localhost:8080 doesn't really work, since the 'view demo' links aren't relative but are to webviz.io) - but this is mostly from guesswork on my part, and it's not at all apparent.

janpaul123, 2019-06-21:
Yeah, documentation for development is super lacking at this point. I at least added a bit more info for how to run in #160. But if you want to just use the tool itself with your ROS bags (and don't care about development) you can just use https://webviz.io/try/ and drag in your bags.

asimay, 2019-06-28:
hi, @janpaul123 , I did as you said, I dragged the bagfile into webviz, but why the camera image cannot show out? I checked the message type is right.

[image]

asimay, 2019-06-28:
and we don't have internet on host-machine, it is setup on car, so how to deploy your tools on our server?

jtbandes, 2019-06-28:
Hi @asimay,

- For your camera image — the Image panel currently doesn't support the `mono8` encoding. The supported `sensor_msgs/Image` encodings are here (note that we also allow `sensor_msgs/CompressedImage`): https://github.com/cruise-automation/webviz/blob/9be1ffe5db4971a991a018ce6413be76a7a18261/packages/webviz-core/src/panels/ImageView/ImageCanvas.js#L88-L91 (And we're going to add the other `bayer_*8` variants very soon.) If you're interested in contributing to Webviz, we would welcome a PR to add support for mono8! Otherwise, we'll take note of this feature request, and we can definitely add a better error message in the Image panel so the problem is more obvious 😉

- For deploying Webviz, currently the best option is `npm run build` as documented in [the readme](https://github.com/cruise-automation/webviz/blob/master/README.md). That will compile the package with webpack, so it can be run with node. It would be possible to, for example, create a Node.js Docker image that runs the server. Alternatively, the JS bundle could be loaded from a static HTML page, served with any simple web server. Feel free to file a new issue if you get stuck with this.

jtbandes, 2019-07-16:
@asimay mono8 and mono16 encodings are supported now as of #174. Give it a try!

Going to close this issue now as we've updated the documentation a little bit and provided some more deployment info. Feel free to open a new issue if you have a specific question or problem with getting it to run.

closed 2019-07-16 by jtbandes