cruise-automation/webviz #162: ouster ros bags not working

rfsch, 2019-06-23:
Hi 

I cannot drop the ouster ros bag files into webviz 
http://data.ouster.io/sample-data-2018-08-29/index.html

thanks

jtbandes, 2019-06-24:
Hi, thanks for your interest in webviz! It looks like these bags each only have two topics, is that right?

[image]

My first guess would be that the topics are not correctly handled because they don't start with a `/`. (Webviz might be making some incorrect assumptions there, because the data we usually work with has universally `/`-prefixed topic names.) As a quick workaround, you can try renaming the topics so they start with `/`.

andre-nguyen, 2019-06-25:
The bags are not handled because the topics are non-standard. They are the raw dumps from the ouster lidar, you have to convert those to point clouds and save the point clouds as bags.

janpaul123, 2020-04-09:
I'm going through some old tickets to clean them up: indeed we currently only support the standard [`sensor_msgs/PointCloud2`](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/PointCloud2.html) message type for point clouds. So just convert the bag to use those and it should work. Let us know if you have any more issues!

closed 2020-04-09 by janpaul123

zlg9folira, 2020-08-27:
I have `/os1_cloud_points/points` of type `sensor_msgs::PointCloud2` in my bag files. I am trying to visualize through 3D panel inside WebWiz. It however does nothing. There is a toggle point next to topic name in the panel. When I select that, a red `error` sign appears:
`Missing transform to root transform map`

Any idea ?

vidaaudrey, 2020-08-27:
Hi @zlg9folira, at the right side of the 3D panel, you could choose which the root frame and all the other markers will be rendered according to the correct root frame. As for why do we need `/tf`, here is the link to read more: http://wiki.ros.org/tf#:~:text=Package%20Summary&text=tf%20is%20a%20package%20that,any%20desired%20point%20in%20time.