cruise-automation/webviz #169: Usability improvements

felixvd, 2019-07-05:
As mentioned in the Discourse, this is an excellent effort. I noted some UX thoughts upon first usage which I just wanted to drop here. They are loose suggestions:

- Close the layout window upon clicking "Apply" if it was successful
- If the layout window is completely empty and "Apply" is clicked, do not display an error message but insert "{}" and close the window silently
- If the layout is empty when a bag is dropped onto the window and the layout is empty, open a default selection of widgets (image, plot, raw message) pre-subscribed to a sensible topic in the bag

The last one came up because it was confusing to drop the bag, but then see nothing happening. I think that will confuse users who are not familiar with ROS especially.

Cheers!

jtbandes, 2019-07-06:
Thank you, these are great suggestions. 😃 It's always interesting to hear about the experience of someone using an app for the first time; as the developer it's easy to miss the small usability issues. These ones should be pretty easy (and, if anyone is interested, we'd happily accept pull requests!)

Don't hesitate to share any further feedback you might have with more usage.

vidaaudrey, 2019-09-11:
@felixvd Pls see the updates below for the UX improvements you suggested:

> Close the layout window upon clicking "Apply" if it was successful. 

Done.

> If the layout window is completely empty and "Apply" is clicked, do not display an error message but insert "{}" and close the window silently

Done. 

> If the layout is empty when a bag is dropped onto the window, open a default selection of widgets (image, plot, raw message) pre-subscribed to a sensible topic in the bag

We opted out since the user might actually want to have empty layout/globalState config. However, if you open webviz in incognito mode or cleared the localStorage, it'll load with the default layout.

Let us know if you have more feedback / ideas!

vidaaudrey, 2019-10-29:
Will close this issue now. Feel free to open another one when you have other suggestions

closed 2019-10-29 by vidaaudrey