cruise-automation/webviz #170: Rosbridge integration

PymZoR, 2019-07-08:
Hi,

Thanks for the amazing work !
In order to use webviz as a real drop-in replacement of RViz, visualizing real-time data would be really useful. As mentioned in https://github.com/cruise-automation/webviz/issues/118, what are your plans on this feature ? How hard would it be to replace rosbag.js by realtime data provided by the rosbridge suite (https://github.com/RobotWebTools/rosbridge_suite) ?
If you have any insights about where to begin, I would be happy to contribute.

jtbandes, 2019-07-08:
Hi Axel, thanks for your interest in this project!

At a high level, the main thing that's required to implement this is an implementation of the Player interface that communicates with rosbridge: https://github.com/cruise-automation/webviz/blob/9528e0a25c99a4c471563a8dc6aef0083db663f2/packages/webviz-core/src/types/players.js#L139-L155

As I mentioned we do have an internal implementation of this, but it's using a custom server rather than the rosbridge server. We could consider pushing the client code to the open-source repo as a starting point, but there would be no server implementation.

Without having looked into it in detail, some challenges I foresee with rosbridge compatibility might be:
- sending the list of available datatypes up front
- sending [receiveTime](http://docs.ros.org/api/roscpp_traits/html/classros_1_1MessageEvent.html#a21d12c15255967761a0078ef29902414) for each message

I'd be interested to learn more about your workflow though. Are you currently using some of the rosbridge suite tools? Can you share some info about your experiences with them?

PymZoR, 2019-07-09:
> sending the list of available datatypes up front

This might not be ideal for a "real time" utilisation, as some projects dynamically create publishers / subscribers. For this, maybe consider to a dynamic stream of datatypes ?

Putting aside this use case, listing the datatypes upfront requires to wait for all nodes to be launched in order to avoid race conditions. This could be done by a node responsible of sending all webviz-specific messages using the roslaunch CLI `roslaunch --nodes` or [API](https://github.com/ros/ros_comm/blob/4833dc8962e1dec03c71475288bcb45edbb22597/tools/roslaunch/src/roslaunch/node_args.py#L67) (undocumented) +  `rostopic list` (we already use this *hacky* solution internally). Once the nodes are all started, and considering all pub/sub are in place, the rosbridge suite already expose this information in it's sub-package  [rosapi](https://github.com/RobotWebTools/rosbridge_suite/tree/develop/rosapi) thourgh the service `/rosapi/topic_type` (see impl [here](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosapi/scripts/rosapi_node#L51) and [here](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosapi/src/rosapi/proxy.py#L51)).

> sending receiveTime for each message

In the bag context, the receiveTime is an important meta-information as it represents the time at which the message has been received by the ROS master during the runtime.
I think that using as receiveTime the stamp at which the browser receive the message (thus delayed by the rosbridge) makes sense as during a live run, you could consider the browser as yet another node (like RViz). Delay induced by the transport layer, as experienced by all nodes, visualization tools included, *expected* and i would say *normal*. Please correct me here if I don't grasp the issue tied to this.

To answer your question, we use extensively the rosbridge_suite as a gateway for our web apps (remote control GUI) and for our unity-based simulator (see [RosSharp](https://github.com/siemens/ros-sharp) project) with great success. It's still far from perfect and we still experience some issues, but the rosbridge_suite folks are reactive whenever we open an issue / pull request :)

janpaul123, 2019-07-23:
> receiveTime the stamp at which the browser receive the message

But when you're using a simulated clock it would make more sense for the receive time that shows up in the UI to correspond to simulated clock values (and subscribing to /clock over a websocket is a bit much). At least that's how we've implemented it in our internal version.

dfeldstarsky, 2019-10-04:
> As I mentioned we do have an internal implementation of this, but it's using a custom server rather than the rosbridge server. We could consider pushing the client code to the open-source repo as a starting point, but there would be no server implementation.

@jtbandes That implementation would be incredibly useful for me and my company. We currently have a data lake that we can play into our simulators (using new rosbags as an intermediary), but also visualization tools that operate sans-rosbags. If you opened up the client code, we could definitely build upon it with our own data store. As it sits now, if we want folks to be able to use web-viz we'd need some way for them to get small snippets of rosbags.

Our use case is different than this issue (rosbridge + webviz to replace rviz) though, and I can open a new issue if you think it worth it.

janpaul123, 2019-10-05:
@dfeldstarsky for your use case, perhaps it would help to use our bag streaming feature plus the seek-to parameter? Like https://webviz.io/try/?remote-bag-url=https://open-source-webviz-ui.s3.amazonaws.com/demo.bag&seek-to=1490150281000 Not sure if that will be sufficient for your use case though?

dfeldstarsky, 2019-10-05:
Yup that would work for the case where we're streaming out of historic bags - however, a growing chunk of our data never makes it to bags in the first place. We directly write it down to our data store, sans ros. 

Internally at cruise are you always serving bags to the webviz frontend? Or are there implementations similar to what we're trying to do at Starsky? 

I'll have to dig through the bag streaming and seeking code, it's really really performant :). Been a joy to use! And this will work for us for now to prove out if we can use webviz in our automated workflows at least.

janpaul123, 2019-10-05:
Internally we typically don't stream from bags either. :) We use Avro files, one for each topic, and then combine them into only the topics needed. Much faster than streaming entire bag files. What do you use exactly?

dfeldstarsky, 2019-10-07:
Our datastore is very similar, subbing in parquet instead of avro. We use Presto (Athena as a managed service  AWS) to query. We could easily stream out the query result, as it's saved as a .csv w/ the Presto architecture.  

Is there a plugin or entrypoint somewhere within the Webviz world we could point to that rather than a bagfile?

janpaul123, 2019-10-07:
Sadly no, our Avro integration is quite complicated and tied to our implementation so I can't point you to something you can easily drop in. However, we're thinking about defining a sort of "streaming bag format" which anyone could potentially write a webserver for. I'll let you know if we end up doing something like that.

dfeldstarsky, 2019-10-07:
Ahh understood. That would be great if you do :).

We're still very new to the whole webviz & worldview ecosystem, so we'll keep exploring and find a way to make it work.

janpaul123, 2019-10-07:
Cool, let us know what you come up with!

MarcoRuzzon, 2019-12-11:
Hi, 
is there any update?  @PymZoR did you manage to implement what you proposed on the 9th of July?

PymZoR, 2019-12-12:
@MarcoRuzzon Sadly no, we do not have resources allocated to this at the moment

janpaul123, 2019-12-27:
@PymZoR @dfeldstarsky @MarcoRuzzon This is now working, see https://github.com/cruise-automation/webviz/pull/299 for details!

janpaul123, 2019-12-27:
cc @varandaas @vinaykumarhs2020 @sromberg @srsidd (I saw that you 👍 this ticket)

closed 2019-12-27 by janpaul123

vinaykumarhs2020, 2019-12-27:
Thanks 😊. Giving it a try soon.