cruise-automation/webviz #173: https://webviz.io/try/ seems broken

shufengh, 2019-07-14:
The tryout website seems broken as it cannot load wasm-lz4 because the link resolves to webviz.io but the resource seems under https://github.com/cruise-automation/wasm-lz4. 

Error chrome devtool console,

```
Uncaught (in promise) abort("abort(\"on the web, we need the wasm binary to be preloaded and set on Module['wasmBinary']. emcc.py will do that for you when generating HTML (but not JS)\"). Build with -s ASSERTIONS=1 for more info."). Build with -s ASSERTIONS=1 for more info.
```
Chrome version
```
Version 75.0.3770.100 (Official Build) (64-bit) for mac
```

janpaul123, 2019-07-21:
I think that's just a warning, which we should eventually fix, but shouldn't prevent you from using the tool. https://webviz.io/try/?demo works fine for me (exact same browser version).

shufengh, 2019-07-27:
Hi @janpaul123, I tried to load a rosbag I downloaded from https://projects.csail.mit.edu/stata/downloads.php and specific file http://infinity.csail.mit.edu/data/2011/2011-01-24-06-18-27.bag. There's no error besides the warning so i wonder what could go wrong.

vidaaudrey, 2019-08-06:
@shufengh  I just tried and the bag is loading for me as well. There might be warnings in the 3D panel about missing transforms which you can get around by clicking the `follow` icon at the top right corner.

[image]

shufengh, 2019-08-11:
Hi @vidaaudrey you are right! Think the default view is to follow velodyne which the bag does not seem to have. Thanks and resolving.

closed 2019-08-11 by shufengh