cruise-automation/webviz #181: Problem with installing/running on Ubuntu 18.04

fkeidel, 2019-07-31:
I have several problems with installing/running WebViz on Ubuntu 18.04
1) I had to install nodejs and npm. I followed the description on 
https://wiki.ubuntuusers.de/Node.js/
When installing npm, I got the error, that dependencies are missing that cannot be resolved (nodejs-dev node-gyp libssl1.0-dev) 
So I followed the hints on https://askubuntu.com/questions/1088662/npm-depends-node-gyp-0-10-9-but-it-is-not-going-to-be-installed
and did 
sudo apt-get install nodejs-dev node-gyp libssl1.0-dev
But this command deinstalled many of the libraries in my system especialle my ros development environment. I am not able to work with the ros tools anymore.

2) After installing nodejs and npm, I followed the guidelines in the github readme.
npm run bootstrap. I remember that there were 2 red lines in the output, that  dependencies could not be resolved. 

3) When starting webviz, the browser open and I got the message that there is no access to cruise-control/button. So I changed in the package directory and installed cruise-control/button separately. But when starting webviz, there was the next message, that popper.js is missing. So I installed all packages in the folder packages separately. Is this correct? Do I have to install them by myself separately?

4) After installing all sub-packages, starting WebViz (http://localhost:8080/try/), only a black background appears. In the console I get the error:
ERROR in ./packages/webviz-core/src/components/withDragDropContext.js
Module not found: Error: Can't resolve 'react-dnd-html5-backend' in '/home/q456941/webviz/packages/webviz-core/src/components'
 @ ./packages/webviz-core/src/components/withDragDropContext.js 10:51-85
 @ ./packages/webviz-core/src/components/Root.js
 @ ./packages/webviz-core/src/loadWebviz.js
 @ ./packages/webviz-core/src/index.js

How to proceed? How to install/run WebViz on my computer (ubuntu 18.04) correctly?

----------------
npm-debug.log
 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'run', 'build' ]
2 info using npm@3.5.2
3 info using node@v8.10.0
4 verbose run-script [ 'prebuild', 'build', 'postbuild' ]
5 info lifecycle root@~prebuild: root@
6 silly lifecycle root@~prebuild: no script for prebuild, continuing
7 info lifecycle root@~build: root@
8 verbose lifecycle root@~build: unsafe-perm in lifecycle true
9 verbose lifecycle root@~build: PATH: /usr/share/npm/bin/node-gyp-bin:/home/q456941/webviz/node_modules/.bin:/opt/ros/melodic/bin:/home/q456941/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/bin:/home/q456941/llvm-project/build/bin
10 verbose lifecycle root@~build: CWD: /home/q456941/webviz
11 silly lifecycle root@~build: Args: [ '-c', 'lerna run build && webpack' ]
12 silly lifecycle root@~build: Returned: code: 2  signal: null
13 info lifecycle root@~build: Failed to exec build script
14 verbose stack Error: root@ build: `lerna run build && webpack`
14 verbose stack Exit status 2
14 verbose stack     at EventEmitter.<anonymous> (/usr/share/npm/lib/utils/lifecycle.js:232:16)
14 verbose stack     at emitTwo (events.js:126:13)
14 verbose stack     at EventEmitter.emit (events.js:214:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/share/npm/lib/utils/spawn.js:24:14)
14 verbose stack     at emitTwo (events.js:126:13)
14 verbose stack     at ChildProcess.emit (events.js:214:7)
14 verbose stack     at maybeClose (internal/child_process.js:925:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
15 verbose pkgid root@
16 verbose cwd /home/q456941/webviz
17 error Linux 4.15.0-55-generic
18 error argv "/usr/bin/node" "/usr/bin/npm" "run" "build"
19 error node v8.10.0
20 error npm  v3.5.2
21 error code ELIFECYCLE
22 error root@ build: `lerna run build && webpack`
22 error Exit status 2
23 error Failed at the root@ build script 'lerna run build && webpack'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the root package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     lerna run build && webpack
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs root
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls root
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

vidaaudrey, 2019-08-06:
Hey @fkeidel, could you try running `npm run build` before `npm run docs`? Basically the code is referring some internal packages. We forgot to put this step in the README, and it'll be updated in #177.  Let me know if you still have issue running the app.

vivekrathi14, 2020-09-21:
@jtbandes Did you solve this issue?

troygibb, 2020-09-21:
@vivekrathi14 this was probably closed due to inactivity-- are you running into the same issue?

vivekrathi14, 2020-09-21:
@troygibb Yes I am facing the same issue. Same problem with npm as per above points 1 & 2. Initially I was able to do  `sudo npm run bootstrap` (I am in webwiz directory while running this). However I got errors with npm run build. I wasn't able to fix them. After doing this when I ran my ros project almost all packages like gazebo, rviz were uninstalled (similar to 1) & later doing `npm run bootstrap` gives error. Also sudo npm doesn't work. I checked `whereis node` and it did show /usr/bin/node.

I didn't do anything else worrying about installing ros packages again.
I am using Ubuntu 18.04, ROS melodic.

inactivist, 2021-02-12:
As an aside, you can avoid using `sudo` by installing and managing Node versions using [nvm](https://github.com/nvm-sh/nvm).  
 That's what I'm doing on Ubuntu 18.04.

There are other ways of installing Node locally (non-sudo) but I've settled on `nvm` for now.