cruise-automation/webviz #184: Plugin Architecture For 3D Display

MicahChambers, 2019-08-05:
It would be nice to have the ability to display any message in the 3D visualizer without producing markers ahead of time similar to Plotting doesn't require a special format. 

Is there any work going on to do this? It is obviously more difficult to write a generic message to message conversion system than simply finding a field to plot. Maybe allowing users to dump a javascript function or class along the lines of:
```javascript
{
   topic: 'xxx',
   convert: function(msg: ThingMsg) {
        ...
        return markers;
   }
}
```
Any thoughts? Thanks for doing great work btw.

vidaaudrey, 2019-08-05:
Hey @MicahChambers, Webviz currently already have such structure called `Node`. You can define a Webviz Node like [this](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/pipeline/nodes.test.js#L105):
``` 
      const NodeB = {
        datatypes: {}, 
        inputs: ["/webviz/a/counter"],
        output: { name: "/webviz/b", datatype: "b" },
        callback({ message }) {
          return {
            messages: [makeNodeMessage("/webviz/b", "b", { count: message.message.count })],
            state: undefined,
          };
        },
      };
```
and register it in [loadWebviz](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/loadWebviz.js#L16), e.g 
```
nodes: () => {
    const NodeB = require("SOME_PATH").default;
    return [NodeB];
}
```
All panels will have access to the node messages. We are also actively working on a feature called Node Playground which will allow people to create Webviz nodes dynamically. Stay tuned!

MicahChambers, 2019-08-05:
Thanks Audey! Thats perfect!

closed 2019-08-05 by MicahChambers