cruise-automation/webviz #191: where to configure fix ip rather than localhost?

david-gwa, 2019-08-16:
hey, webviz team,

        not familiar with web app, a simple wonder, where to setup a fixed IP? 

        Thanks

jtbandes, 2019-08-16:
What commands are you currently using / what exactly are you trying to do?

Maybe you're interested in configuring some host settings when using the webpack dev server: https://github.com/webpack/webpack-dev-server/issues/147#issuecomment-89895948

Or are you more interested in packaging/deploying the app somewhere, in which case you might find this discussion useful: https://github.com/cruise-automation/webviz/issues/161

david-gwa, 2019-08-16:
thanks, really first time to run webpack:
after  npm run build
I can do: 
webpack-dev-server.js --host 10.20.1xx.1xx --port 8xxx

and when access in browser, it 's the host ip now.

closed 2019-09-10 by jtbandes