cruise-automation/webviz #192: Getting click location with good performance

conrad, 2019-08-16:
Hello, I'm trying to build a(n) UI with point-and-click navigation, and in order to do so right now I'm trying to draw many invisible polygons across the floor, but the performance drop is making it unusable :(

Here's what I'm writing:
```
  getFloor = () => {
    const floorIncrementM = 0.5  // meters
    let sectionId = 1111
    let floorPolygonData = []
    for (let i = -FLOOR_LENGTH; i < FLOOR_LENGTH; i = i + floorIncrementM) {
      for (let j = -FLOOR_LENGTH; j < FLOOR_LENGTH; j = j + floorIncrementM) {
        floorPolygonData.push(
          {
            id: sectionId++,
            coordinates: { x: i + floorIncrementM / 2, y: j + floorIncrementM / 2 },
            points: [
              { x: i, y: j, z: 0 },
              { x: i, y: j + floorIncrementM, z: 0 },
              { x: i + floorIncrementM, y: j + floorIncrementM, z: 0 },
              { x: i + floorIncrementM, y: j, z: 0 }
            ],
            color: WORLDVIEW_RGB.clear,
          }
        )
      }
    }

    return (
      <FilledPolygons
        onClick={this.handleFloorClick}>
        {floorPolygonData}
      </FilledPolygons>
    )
  }
```

conrad, 2019-08-16:
What I'd like to write looks like this:
```
  getFloor = () => {
    const vertices = [
      { x: FLOOR_LENGTH / 2, y: FLOOR_LENGTH / 2, z: 0 },
      { x: -FLOOR_LENGTH / 2, y: FLOOR_LENGTH / 2, z: 0 },
      { x: -FLOOR_LENGTH / 2, y: -FLOOR_LENGTH / 2, z: 0 },
      { x: FLOOR_LENGTH / 2, y: -FLOOR_LENGTH / 2, z: 0 },
    ]
    const floorPolygonData = [
      {
        id: 1111,
        points: vertices,
        color: WORLDVIEW_RGB.clear,
      }
    ]
    return (
      <FilledPolygons
        onClick={this.handleFloorClick}>
        {floorPolygonData}
      </FilledPolygons>
    )
  }
```

In this, the click event would have to get the coordinates of where on the object the click hit. Is there a way to do this?

vidaaudrey, 2019-08-18:
Hey @conrad  the `onClick` event does give you a `ray` object which contains the raycasting information including dir (direction), origin, and clicked point (all in Vec3 format). Try it out and see if satisfies your requirement.

Regarding on the performance, there is some extra computation in `FilledPolygons` to get the points. You could try use `Triangles` directly. https://github.com/cruise-automation/webviz/blob/master/packages/regl-worldview/src/commands/FilledPolygons.js#L71 

In addition, Worldview `0.2.0` is **unofficially** available now (ref https://github.com/cruise-automation/webviz/pull/177). It has some performance optimizations and you no longer need to pass the `id` in. We've been testing it internally with lots of moving objects and PointCloud. Let us know if you have any questions or still see perf drops after you try it out.