cruise-automation/webviz #197: Issues with decoding .glb files

spjy, 2019-08-19:
I am attempting to load .glb files other than the Duck and CesiumMan that are provided in the example. I have tried to import my own models as well as ones provided in the example model repository [here](https://github.com/KhronosGroup/glTF-Sample-Models/tree/master/2.0) such as `Box.glb` and `RiggedSimple.glb`, but I always get the error "Error decoding GLB file".

I have forked your GLTFScene example and tried importing my own model to it, namely `rectangle.glb` and I get the same error as above.

Example: https://codesandbox.io/s/agitated-star-ruxr4?fontsize=14

jtbandes, 2019-08-22:
Thanks for the report. This looks like a regression caused by #198; we'll look into it soon. cc @davidswinegar

jtbandes, 2019-09-10:
Tested out your CodeSandbox again and it looks like it's working now, after David's fix:

[image]

closed 2019-09-10 by jtbandes