cruise-automation/webviz #204: A problem to watch webviz demo, player is always initializing

masakisonic, 2019-08-22:
Hi, I have a problem on watching webviz demo from [https://webviz.io/try/?demo](https://webviz.io/try/?demo), 
**The player always waits for initializing**, so I can't watch the demo.

I use Chrome on Windows 7 and its version
> 76.0.3809.100 (Official Build) (64-bit)

I also try my phone & ubuntu 16.04, but the same result for me.

Does anyone know if I lose some plugins or other things on watching demo?
Thanks.

[image]

janpaul123, 2019-08-22:
Oh you're right, thanks for reporting, we'll look into it!

troygibb, 2019-08-22:
It should be fixed now!!! Sorry for the inconvenience.

closed 2019-08-22 by troygibb