cruise-automation/webviz #205: Wrapping an element around WorldView component causes infinite increase in height

spjy, 2019-08-22:
Whenever I wrap an HTML element around the WorldView component, it infinitely increases its height. I am forced to set a fixed height and width as a result.

See https://codesandbox.io/s/relaxed-sun-tqxk3.

vidaaudrey, 2019-08-22:
Hi @spjy  that's a known issue. For now, you could try applying `height: "100%" ` to the Worldview wrapper div to make sure by default Worldview fills up the whole area. 
https://codesandbox.io/s/quizzical-kalam-ducme

spjy, 2019-08-22:
Right, that is what I am doing now; however that is a band aid solution and is also impractical in my case as I have a wrapper component that requires block level elements.

jtbandes, 2019-08-22:
Can you describe more about how you expect the Worldview component should be sized in your case? Is something constraining its size from the parent down, or how do you expect it to choose its size?

(FYI the issue is a side effect of how we're using [react-container-dimensions](https://www.npmjs.com/package/react-container-dimensions) inside Worldview)

spjy, 2019-08-24:
I expect it to be sized based on its immediate parent container.