cruise-automation/webviz #209: visualize data realtime

PatrickQianKun, 2019-08-24:
Hi, 
thanks for your work!

For real-time usage, I plan to gather data from radar, lidar or camera through CAN, Ether, etc.. use backend to synchronize and pack these data in ROS like messages, then send them to webviz through websocket directly to each message consumers. 

Any suggestions about this. I really appreciate it if you guys could give me some help.

jtbandes, 2019-08-24:
Thanks for your interest :) I would recommend you read #118 and #170, we have already discussed live connections there.

janpaul123, 2019-12-27:
@PatrickQianKun This is now working, see https://github.com/cruise-automation/webviz/pull/299 for details!

janpaul123, 2019-12-27:
cc @beetleskin also (I saw that you 👍 this ticket)

PatrickQianKun, 2019-12-27:
👍 👍 👍