cruise-automation/webviz #225: How to set backgroundImage for scene?

artvichi, 2019-09-09:
Hi! Is it possible to add a .jpeg image to a scene? (I saw you have added road and other images in your blog) .
Also, is there a way to embed OccupancyGrid ?

jtbandes, 2019-09-09:
Are you interested in a fixed background image, or an image that appears at a certain position in space (and moves with the camera)? For our map, we created a new Worldview command that loads and renders the images.

Are you using Webviz or are you creating your own React app with Worldview?

artvichi, 2019-09-10:
@jtbandes , thanks for fast response!
 Worldview `regl-worldview` repo is used. Tried to do so with `Overlay` and `GTLFScene` but I hoped you have some simplest way to do that... 
I need set a floor picture of the place where my robot drives. Image that may appear in different locations would be better. Ideally would be a possibility to add few layers on the map (but it's advanced feature).

jtbandes, 2019-09-10:
We don't have any built-in image renderer, but you can make one yourself with a custom command. Here's an example to get you started: https://codesandbox.io/s/nervous-elbakyan-0215j

artvichi, 2019-09-10:
@jtbandes , thanks a lot!

closed 2019-09-10 by artvichi