cruise-automation/webviz #226: MouseEvent arguments mix-up

conrad, 2019-09-10:
Hello, I've just upgraded to v0.2.2, and two of the fields seem to have been swapped unintentionally. Previously, the arguments were `event, { ray, object(s) }`. 

Now, I'm seeing that the second argument is still an object with the field ray, but this field then points to a mouse event object. Meanwhile, the first argument sent turns out to be a ray. To explain a little more this what the second argument looks like now: 
```
{
  objects: [{…}]
  ray: Class
    altKey: false
    bubbles: true
    button: 0
    buttons: 0
    cancelable: true
    clientX: 640
    clientY: 233
    ctrlKey: false
    ...
  }
}
```

That ray looks a lot like a mouse event...

I'm guessing this was just an accident. I'll switch these out in my code, in the meantime, and leave myself a note.

conrad, 2019-09-10:
Or I'll dig around and make a PR if I find the issue.

davidswinegar, 2019-09-10:
Thanks for the bug report! I'll take a look at this today.

PR #227 (2019-09-11): Fix mouse event argument mix-up
## Summary
Fix bug of switching event and ray in object callbacks. Fixes #226.

## Test plan
Included integration test.

## Versioning impact
Upgraded to version 0.2.3.
commit 71a5c13a (2019-09-11): Fix mouse event argument mix-up (#227)

Fix bug of switching event and ray in object callbacks. Fixes #226.

Test plan: included integration test.
closed 2019-09-11 by davidswinegar