cruise-automation/webviz #230: Cannot access webiz through browser and tests are failing

gismo07, 2019-09-20:
I wanted to test webviz locally, therefore I pulled the latest version and tried to open the web app.

What I did so far:
```
apt update
apt install git nodejs npm curl
npm install -g npm 
npm install -g n
n latest
npm install node-sass  

cd webviz
npm run bootstrap 
npm run build
```

If now the tests are performed, we get the following error messages:
```
Summary of all failing tests
 FAIL  packages/webviz-core/src/players/RandomAccessPlayer.test.js (13.188s)
  ● RandomAccessPlayer › backfills previous messages on seek

    Timeout - Async callback was not invoked within the 5000ms timeout specified by jest.setTimeout.

      450 |   });
      451 |
    > 452 |   it("backfills previous messages on seek", async () => {
          |   ^
      453 |     const provider = new TestProvider();
      454 |     const source = new RandomAccessPlayer({ name: "TestProvider", args: { provider }, children: [] });
      455 |     let callCount = 0;

      at Spec (node_modules/jest-jasmine2/build/jasmine/Spec.js:85:20)
      at Suite.it (packages/webviz-core/src/players/RandomAccessPlayer.test.js:452:3)
      at Object.describe (packages/webviz-core/src/players/RandomAccessPlayer.test.js:133:1)

  ● RandomAccessPlayer › discards backfilled messages if we started playing after the seek

    Timeout - Async callback was not invoked within the 5000ms timeout specified by jest.setTimeout.

      537 |   });
      538 |
    > 539 |   it("discards backfilled messages if we started playing after the seek", async () => {
          |   ^
      540 |     const provider = new TestProvider();
      541 |     const source = new RandomAccessPlayer({ name: "TestProvider", args: { provider }, children: [] });
      542 |     let callCount = 0;

      at Spec (node_modules/jest-jasmine2/build/jasmine/Spec.js:85:20)
      at Suite.it (packages/webviz-core/src/players/RandomAccessPlayer.test.js:539:3)
      at Object.describe (packages/webviz-core/src/players/RandomAccessPlayer.test.js:133:1)

 FAIL  packages/webviz-core/src/util/debouncePromise.test.js
  ● debouncePromise › debounces with resolved and rejected promises

    expect(received).toBeUndefined()

    Received: {}

      32 |     await Promise.resolve();
      33 |     expect(calls).toBe(2);
    > 34 |     expect(debouncedFn.currentPromise).toBeUndefined();
         |                                        ^
      35 |
      36 |     debouncedFn();
      37 |     expect(calls).toBe(3);

      at Object.toBeUndefined (packages/webviz-core/src/util/debouncePromise.test.js:34:40)

Test Suites: 2 failed, 73 passed, 75 total
Tests:       3 failed, 748 passed, 751 total
Snapshots:   0 total
Time:        42.682s
```

I'll think the proper way of running the web app is to execute `npm run docs` or?  However, the output of the browser at either localhost:8080 or localhost:8080/try is empty.

jtbandes, 2019-09-23:
Hi, can you verify which version of node you're using? We usually use v10 — and yes, `npm run docs` is the right way to go.

surajhpatil, 2019-09-26:
Currently it's failing for me (locally as well as your try demo page) with following issues with fonts?

[image]

After this nothing happens and it just stops/crash.
Additional console log

[image]

jtbandes, 2019-10-18:
That's interesting; which version of Chrome are you using and on what platform?

surajhpatil, 2019-11-26:
Sorry for the late response. I am using chrome `Version 77.0.3865.90 (Official Build) (64-bit)`.
I figured this out that one of the plugin installed in my browser was causing this issue somehow (fonts related to that plugin).
Currently I have bypassed the waitForFont method so that I can at-least load the webviz.

jtbandes, 2019-11-26:
This ticket is pretty old and I'm not sure anything more is needed from us here, so I'm going to close it — feel free to file a new ticket with more information if you still need help.

closed 2019-11-26 by jtbandes