cruise-automation/webviz #248: docker images for webviz

stefan-enway, 2019-10-07:
I tried building via the Dockerfile, is there a prebuilt image somwhere?
what options would i use to launch it, if I had build it?

janpaul123, 2019-10-07:
We currently only use the Dockerfile in CI to run our tests; see [here](https://github.com/cruise-automation/webviz/blob/522cb4400c5c42220e07b3f46854087da75a24b5/.circleci/config.yml#L28). It's available on dockerhub under cruise/webviz-ci. We currently don't use it for our application, but instead we generate a static hosted website using [`npm run docs-deploy`](https://github.com/cruise-automation/webviz/blob/522cb4400c5c42220e07b3f46854087da75a24b5/package.json#L131). Hope this helps!

closed 2019-10-07 by janpaul123

stefan-enway, 2019-10-07:
it does. I am late to the whole JS/node/react thing and was expecting a port or something. Thanks!

janpaul123, 2019-10-07:
Yeah, it would be cool for us to publish all of this under a proper Docker image some day!

stefan-enway, 2019-10-08:
We will probably use Webviz so maybe i can provide some PRs in a couple of weeks. How do you handle all this dependacy-hell-PRs ? I tried to "just" merge them but that does only work in the right order and sometimes just dont work if newer version change their API.

janpaul123, 2019-10-09:
Ah yeah we should probably actually pay attention to Dependabot. Don't worry about it for now!