cruise-automation/webviz #253: node js fun

stefan-enway, 2019-10-18:
hi, we are tinkering with the webviz to run and build in docker, did you stumble about this error?

```
gyp WARN install got an error, rolling back install
gyp ERR! configure error 
gyp ERR! stack Error: unexpected end of file
gyp ERR! stack     at Zlib.zlibOnError [as onerror] (zlib.js:154:17)
gyp ERR! System Linux 4.15.0-55-generic
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /node_modules/weak-napi
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/webpack-dev-server/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/jest-haste-map/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

```

jtbandes, 2019-10-18:
I've never seen this error before; `unexpected end of file` sounds like it might be a download error, are you sure the network connection is working properly?

stefan-enway, 2019-10-19:
yes, it has because it downloaded a bunch of other libs. I think my colleague did some node.js version upgrades. As you recommended, we should not tinker with that (the good old node.js/rubygems/python/{yourscriptinglanguagehere} dependency hell), so we will probably roll back that one. But we will need a Github Action/Jenkins job (don't laugh) that builds webviz from the scratch. Its on my to do for next week.

closed 2019-11-05 by stefan-enway

jtbandes, 2019-11-05:
Did you need any more help on this or did you get it to work?