cruise-automation/webviz #254: split rosbag during webviz running

david-gwa, 2019-10-24:
hey, cruise team,

       thanks for this great tool, wonder is there an easy way to split the inputted ros-bag at some certain timestep(maybe a start time and an end time during playing the rosbag in webviz), where maybe the testers are very interested at.

     david

janpaul123, 2019-11-06:
Currently this is not possible from the Webviz UI, but you can use [`rosbag filter` for this](https://answers.ros.org/question/99711/how-to-split-a-recorded-rosbag-file/?answer=234004#post-id-234004). The easiest way right now to get those timestamps is to open the Raw Messages panel on a frequent topic and getting the seconds/nanoseconds from there.

janpaul123, 2019-12-31:
Btw, we added some documentation on how to stream a bag from cloud hosting, and that also includes a `seek-to` URL parameter so you can have Webviz navigate directly to the relevant timestamp. I hope that's also helpful. Going to close this ticket for now.

closed 2019-12-31 by janpaul123