cruise-automation/webviz #256: Tagging feature in webviz

kissofwind, 2019-10-28:
Hi,  Thanks for the amazing work !
I am new to webviz, but I can feel it is a very promising project. It makes exploring rosbag data much easier than using the tools in ros. But what I feel is just viewing data in rosbags is the first step, after viewing the data, users may want to add comments or tags to this bag file or to part of the data in the bag file.  So I wander if you plan to add some labeling/tagging features to webviz? Say there is cutin scenario in this rosbag file , and I want to record this information and later I want to replay this cutin  part in this rosbag file . Such feature will make webviz more powerful to AD engineers. 
Do you have such plans ?

jxjshaw, 2019-10-28:
Love the idea. You also mention an idea around tags - curious how this might play out in your current workflow involving rosbags? For example do you work with bags of longer duration, so it’s easy to loose track of notable occurrences within the timeline?

kissofwind, 2019-10-28:
what I am thinking of is if webviz can support tagging, for example if there is a cut-in scenario some where in the bag file, then I can create tag info like type of scenario, start time of scenario, end time of scenario, speed of ego vehicle etc. to a tag file or I can even save these tag info along with the path of bag file into a database system. then  i can later search the cut-in scenarios in the database system and find the rosbag file and then replay the bag file from the start time info in the tag file. 
this is a simple example, but consider if there are over PBs of data, using tag info will significantly improve our efficiency in managing these rosbag files.

jxjshaw, 2019-10-28:
Agreed. How do you currently manage these rosbag files? Wondering if there are any existing solutions out there for easy integration

kissofwind, 2019-10-29:
i am using an open source [https://github.com/swri-robotics/bag-database](https://github.com/swri-robotics/bag-database) tool to manage the raw rosbag files, but I am also lacking the tagging tool. But I see the potential in webviz and i hope webviz can work as a tagging tool.

janpaul123, 2019-11-06:
This is a great idea and internally also an often requested feature. It sounds like you mostly want to put interesting information along the timeline, is that correct? E.g.:
- 10 seconds: "interesting scenario starts here #cutin"
- 13 seconds: "note that the speed of the vehicle that cuts in is 20kph"
- 18 seconds: "scenario ends"

It would be cool to allow putting those "tags" or "labels" directly on the timeline, and then export them to a JSON file or so (and import them again by dragging in such a JSON file). Is this kind of how you're thinking about it too?

kissofwind, 2019-11-12:
yes exactly! this is the key feature we want to have because the bag file could be like GBs but only like seconds or minutes of data is useful. and because bag files are unstructured data, Supporting tagging can make it possible to build a sql database system to manage the tags and the bag files better. 
In my understanding as cruse-automation is doing a world class ads development, you must have a lot of ads data collected, I am also very curious how do you manage those data and use those data in a efficient way?