cruise-automation/webviz #260: Switching between controlled & uncontrolled camera views

conrad, 2019-10-29:
I have an application where I'm trying to switch between having the user control the camera position and having the automatically camera follow vehicles in the visualization. However, when I switch out using the property cameraState & defaultCameraState, the screen goes blank. Each of these renders on its own, but when I try to change between the two--going either way from one to the other--Worldview dies without an error. Here is the code from the render method of a React container component:
```
          {this.props.map.cameraFollow.shouldFollow ?
            <Worldview
              cameraState={this.cameraState}
              hitmapOnMouseMove
              backgroundColor={[0.4, 0.4, 0.4, 1]}
              onMouseDown={this.onContainerMouseDown}
              onMouseUp={this.onContainerMouseUp}
              onMouseMove={this.onContainerMouseMove}
              onClick={this.onContainerClick}
              onDoubleClick={this.onContainerDoubleClick}>
              {this.getWorldviewElements()}
            </Worldview>
            :
            <Worldview
              defaultCameraState={this.cameraState}
              hitmapOnMouseMove
              backgroundColor={[0.4, 0.4, 0.4, 1]}
              onMouseDown={this.onContainerMouseDown}
              onMouseUp={this.onContainerMouseUp}
              onMouseMove={this.onContainerMouseMove}
              onClick={this.onContainerClick}
              onDoubleClick={this.onContainerDoubleClick}>
              {this.getWorldviewElements()}
            </Worldview>
          }
```

Here I keep all of the values the same except for the `shouldFollow` boolean, and the screen goes blank. Is there something wrong?

vidaaudrey, 2019-10-29:
Hi @conrad, when you are using Worldview as controlled component, you'll need to supply a `onCameraStateChange` handler.  Could you provide more details on the error, e.g. screenshot or codesandbox to reproduce the issue. It's good to refer some examples here https://webviz.io/worldview/#/docs/api/camera

conrad, 2019-11-01:
Thank you. Adding that property (currently, just with a no-op function) has solved this.

closed 2019-11-01 by conrad