cruise-automation/webviz #262: GPS/IMU Topics unavailable/not shown

zlg9folira, 2019-10-30:
Hi, 
I am trying to visualize `/points_raw` as well GPS/IMU topics such as `/imu`, `/fix` or `/gps` in different panels. However, I am able to see and select only `/points_raw` from the topics recorded and exist in bag file. 

As all the topics are published under standard format/name, can you suggest any workaround ?

janpaul123, 2019-11-06:
What are the datatypes of those topics? We currently only support a [handful of datatypes in the 3d panel](https://github.com/cruise-automation/webviz/blob/d01b28e51ed3c39e07c8675825ac536165c9782d/packages/webviz-core/src/panels/ThreeDimensionalViz/SceneBuilder/index.js#L391). You might need a ROS node to convert your data to a supported format, either on your robot or as a post-processing step.

jtbandes, 2019-11-12:
To clarify, we expect _all_ topics in your bag to be viewable with the Raw Messages panel, at least. Are you able to see them there?

zlg9folira, 2019-11-12:
@jtbandes  @janpaul123  I finally figured that out. What I needed was to explore my topics using `plot` panels. I was initially on a wrong panel and that was the reason I was not able to see my topics. Thanks for the response.

closed 2019-11-12 by zlg9folira

jtbandes, 2019-11-12:
Glad it works for you, let us know if you have any further questions!