cruise-automation/webviz #281: Regression in bag playback

iwanders, 2019-12-03:
Hi, thanks for open sourcing webviz!

We've noticed a regression in bag playback. For certain bags with gaps in them the playback halts at the start of the gap instead of continuing playback through them:
[image]

A minimal bag that reproduces the issue on https://webviz.io/try/ : [playback_issue_minimal.bag.gz](https://github.com/cruise-automation/webviz/files/3918713/playback_issue_minimal.bag.gz)
Created with the following script: [make_bag.py.gz](https://github.com/cruise-automation/webviz/files/3918716/make_bag.py.gz), basically it has 2 seconds of data, then a gap of two seconds, and then 30 seconds more data. If the last data part is 29 seconds the playback / cache mechanism jumps over the gap.

I think this was introduced in #273, commits before this don't have this issue. Looking further at the changes in this PR I identified the [these](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/dataProviders/MemoryCacheDataProvider.js#L31-L36) values to affect the behaviour. Changing `BLOCK_SIZE_NS` to something larger (say `2.0e9`) makes the playback work as expected.

@janpaul123 commented in the code that the numbers were determined emperically for good performance. So far increasing the block size doesn't seem to affect the performance too much, but I don't know exactly how the caching mechanism works or how this actually prevents the issue from manifesting.

janpaul123, 2019-12-03:
Thanks for reporting this and the excellent test case. I'll release a fix shortly.

janpaul123, 2019-12-03:
(If you're interested, this line was the culprit: https://github.com/cruise-automation/webviz/pull/283/files#diff-ff549a830b60d70e1ed51ce42222356fL85-R85)

closed 2019-12-04 by janpaul123

iwanders, 2019-12-04:
> If you're interested, this line was the culprit

Ah yes, I see, thanks for linking that.

Really appreciate the quick response and resolution @janpaul123 :+1:  . Confirmed the issue is resolved in the latest `master`.