cruise-automation/webviz #288: Add capability to pass FilledPolygons an array of points arrays for many polygons?

conrad, 2019-12-13:
We're enjoying using this library a lot and hope to share how it's helping us run a fleet of robots. So, thank you :)

Something that could help our performance in the browser would be to allow setting an array of polygon points arrays for the `points` property in `FilledPolygons`. Our systems send us larger arrays of points for many polygons all of the same type or class, so it would be really helpful to be able to simply pass that array of arrays to the points property on one data object. Would that be possible?

vidaaudrey, 2019-12-16:
Hi @conrad, the `instanced` FilledPolygons is a nice feature to have. Should be straightforward to change the logic [here](https://github.com/cruise-automation/webviz/blob/master/packages/regl-worldview/src/commands/FilledPolygons.js#L51-L64) to accept array of points and create triangles accordingly. We might not have the capacity to get to this until early January.  Feel free to send a PR if you are interested in implementing it :)

conrad, 2020-01-10:
Thanks, @vidaaudrey! We also haven't had a lot of bandwidth, but I'll be able to take a look soon.