cruise-automation/webviz #293: Colors switched for rgb pointcloud

shaeqahmed, 2019-12-18:
Pointcloud viewer seems to be interpeting single 24bit value rgb channel/field as bgr.

Sample rosbag here: https://vision.in.tum.de/data/datasets/rgbd-dataset/download#freiburg3_cabinet

Relevant code block: 
https://github.com/cruise-automation/webviz/blob/2e7db3aafffec39b541728668c97ce7d83eee007/packages/webviz-core/src/panels/ThreeDimensionalViz/commands/Pointclouds/PointCloudBuilder.js#L174

Screenshot:
[image]

janpaul123, 2019-12-19:
cc @brianc @vidaaudrey @jtbandes

jtbandes, 2019-12-20:
Thanks for the report. I'm able to reproduce the issue and confirmed webviz's color handling differs from rviz.

jtbandes, 2019-12-20:
I suspect this has to do with our endianness handling. We don't use the `is_bigendian` field, but then again, rviz also seems to ignore it.

References:
https://github.com/ros-visualization/rviz/blob/070835c426b8982e304b38eb4a9c6eb221155d5f/src/rviz/default_plugin/point_cloud_common.cpp#L191
https://github.com/ros-visualization/rviz/blob/070835c426b8982e304b38eb4a9c6eb221155d5f/src/rviz/default_plugin/point_cloud_transformers.cpp#L366

janpaul123, 2020-03-26:
@shaeqahmed Update: it looks like someone internally is going to look into this soon. Btw, I'm trying to get to know our open source community better, and would love to learn more about what you use Webviz for. Could you perhaps shoot me an email at jp.posma@getcruise.com?

janpaul123, 2020-04-09:
This should be fixed now in #396. Let me know if you still see this issue!

closed 2020-04-09 by janpaul123