cruise-automation/webviz #297: Getting error on initalizing webviz

cjds, 2019-12-25:
I tried to build webviz using `npm run build` and `npm run docs` but I get 

```
Error initializing player
Error: Expected 1 top level type definition for 'Header' but found 0
    at findTypeByName (https://webviz.io/dist/WorkerDataProvider.worker.js?0441abdf8c8d84b2f526:122:2477)
    at https://webviz.io/dist/WorkerDataProvider.worker.js?0441abdf8c8d84b2f526:122:3089
    at Array.forEach (<anonymous>)
    at https://webviz.io/dist/WorkerDataProvider.worker.js?0441abdf8c8d84b2f526:122:3027
    at Array.forEach (<anonymous>)
    at parseMessageDefinition (https://webviz.io/dist/WorkerDataProvider.worker.js?0441abdf8c8d84b2f526:122:2977)
    at https://webviz.io/dist/WorkerDataProvider.worker.js?0441abdf8c8d84b2f526:527:333
    at Array.forEach (<anonymous>)
    at exports.bagConnectionsToDatatypes (https://webviz.io/dist/WorkerDataProvider.worker.js?0441abdf8c8d84b2f526:527:278)
    at exports.default.initialize (https://webviz.io/dist/WorkerDataProvider.worker.js?0441abdf8c8d84b2f526:407:3325)
```

when I drag and drop a bag into the UI

cjds, 2019-12-25:
Commit I'm trying out is https://github.com/cruise-automation/webviz/commit/e9835e581cd5e82c5d555d7409bbe79702e7d1a8

janpaul123, 2019-12-26:
Looks like an issue with the bag. How was it recorded? Any chance you can share it with us?

cjds, 2019-12-26:
I'll check but it might have been through a simulator. I will sanitize the bag of private information and attach it

cjds, 2020-01-02:
https://drive.google.com/file/d/1aapnmnWaEPQ0piEowbSGMsoDNmzXVpio/view?usp=sharing

janpaul123, 2020-01-02:
What version of ROS are you on? The message definitions looks pretty old and malformed. First it uses the old `Header` definition instead of `std_msgs/Header`, but that's allowed per the bag format. But also there is a topic which uses `DiagnosticStatus` in the message definition, but the definition for `DiagnosticStatus` itself is not present in that header. That looks malformed per the [description here](http://wiki.ros.org/ROS/Connection%20Header): "message_definition: full text of message definition (output of gendeps --cat)".

cjds, 2020-01-02:
Sorry, I don't think I follow your comment (not super well versed with the rosbag format).

1. We're using Melodic
2. To generate the `DiagnosticStatus` messages we're using https://github.com/ros/diagnostics which hasn't been changed since Indigo but does have a melodic release (maybe a PR to that is needed). (Assuming you are talking about the `/diagnostics_toplevel_state` topic)
3. Could you point me to the topic using the old school `Header` message because I wasn't able to find it using old school `rosbag info`?

Thanks

janpaul123, 2020-01-02:
Huh, weird. To clarify: the problem seems to be that the bag does not properly contain the message definitions inside. How did you generate the bag? Did you just use `rosbag record` or something else?

janpaul123, 2020-01-02:
I suspect that there's something weird with how the bag was generated, because when I run `rosbag fix 2019-12-06-05-44-04.088091_task-3715_action-57980_freight100-1274_0.bag out.bag --force` then it produces a bag that I can play with Webviz just fine.

cjds, 2020-01-02:
I think that may be it...

We use the `rosbag` C++ API directly.

https://github.com/ros/ros_comm/tree/melodic-devel/tools/rosbag

Specifically https://github.com/ros/ros_comm/blob/melodic-devel/tools/rosbag_storage/include/rosbag/bag.h#L165 that line is what it comes down to.

I think I'll check the changes that fix is doing and either make an upstream change to `rosbag` or change our downstream use of it. Thanks for the help.

closed 2020-01-02 by cjds