cruise-automation/webviz #298: Memory leak when use <GLTFScene>

baoyule-js, 2019-12-26:
i use an array to store <GLTFScene>,and refresh the array 10 times per second ,and i find the memory usage exploded.i do not know why.

janpaul123, 2019-12-26:
It would be helpful if you could create a simple webpage that demonstrates the issue; would that be possible?

PatrickQianKun, 2019-12-27:
same issue，use multiple GLTFScene with glb models will cause regl context lost

haojiaping, 2020-10-21:
Hi,Can you speak Chinese?   I found a serious one in your project.

vidaaudrey, 2020-10-21:
Thanks for reporting.  @haojiaping  pls feel free to set up a Google hangout to review/repro the issues together. audrey.li@getcruise.com. Or we could communicate via webviz slack https://join.slack.com/t/webvizchat/shared_invite/zt-hyo9sacz-4jAA5ADjaTaolUBwesMMZQ.

haojiaping, 2020-10-22:
I have sent an email to you. This memory leak problem occurs every time you open a project. After about 10 minutes, the browser's memory will be full and cause a crash.

hhsaez, 2020-11-23:
@baoyule-js @haojiaping @vidaaudrey I've created [this branch](https://github.com/cruise-automation/webviz/pull/554) in order to attempt to reproduce this problem, without luck so far. 

The story I created (executed by using `npm run storybook`) creates a scene with multiple `<GLTFScene />` elements and then starts updating them multiple times. I let it run for about an hour and in my case, memory usage is constant (somewhere in between 220MB and 250MB).  

The question I have is whether or not this story is representative of the scene you're trying to execute. Feel free to modify that code as you see fit in order to reproduce the problem.