cruise-automation/webviz #304: How do I build this to host on my own server?

rgov, 2019-12-31:
I'm trying to build webviz-core so that I can host it on my own web server, where it will be served out of a different path. Here's what I do inside a Docker container:

```
npm run bootstrap
npx lerna run build
npx webpack --env.NODE_ENV=production --config /webpack.config.js
```

These commands are based on the repository README and what the package.json `build` script does.

My `/webpack.config.js` file overrides the project's in a few ways:

```
var config = require("/webviz/webpack.config")

// only build the Webviz core bundle
const entry_target = "webvizCoreBundle"
const entry = config.entry[entry_target]
config.entry = {
    entry_target: entry
}

// override the path on the server
config.output.publicPath = "/static/webviz/"

// re-export the modified config
module.exports = config
```

I take all of the produced files in `./webviz/docs/public/dist/` and copy them to my static files directory.

However, when I navigate to this page in Chrome, I see a lot of noise that it is trying to connect to a local websocket port, and then when I drop a rosbag on it, it tries to load files from `/dist/`, which isn't the path I configured:

    GET http://localhost/dist/f923f2a26c562dd21f5b.worker.js 404 (Not Found)

janpaul123, 2019-12-31:
It might be useful to look at the [`npm run docs-deploy`](https://github.com/cruise-automation/webviz/blob/e57b3a17f6caafd2d152696c220a6c24888f83e9/package.json#L133) command. It puts all files (after having run `npm run build`) into a temporary directory which is then published to our [`gh-pages` branch](https://pages.github.com/), but you can also just statically host that directory yourself.

I'd first get that working before messing with webpack configs.

rgov, 2019-12-31:
Thanks. But the gh-pages branch also hosts its files in the default `dist/` subdirectory. I should be able to override that by replacing the `output.publicPath` setting, but it does not work. Perhaps there is somewhere else in the code where it is hardcoded?

janpaul123, 2019-12-31:
Possibly. I would recommend starting with a working setup, and then making small changes, making sure that nothing breaks in the meantime.

rgov, 2019-12-31:
I believe I can build the demo website OK, the question is more about how to deploy it on sites other than the demo site, with different URL layout, and not re-using the same example HTML pages. A lot of this repo is, for instance, documentation about Worldview,  which I don't want to deploy to my server.

janpaul123, 2019-12-31:
Yeah, we intend for people to use webviz.io/app, so if you want to build your own custom version of Webviz then for now you'll have to figure that stuff out yourself. I'd be interested in learning about the reasons for why you want to customize Webviz, and see if we could instead add features to the tool that everyone would benefit from. But for now I'll close this ticket since customization is an unsupported use case (I'll make a PR to clarify that in the docs a bit better).

closed 2019-12-31 by janpaul123

rgov, 2019-12-31:
I have a web interface to a library of data collected from AUVs. Right now it mostly provides upload, download, and search capabilities. I'd like to embed Webviz for visualization and ideally replace drag and drop with the ability to select a file from the database.

janpaul123, 2019-12-31:
Ah I see. What about linking out to `webviz.io/app`? You can currently stream bags and set layouts and such through the URL ([example](https://webviz.io/app/?remote-bag-url=https%3A%2F%2Fopen-source-webviz-ui.s3.amazonaws.com%2Fdemo.bag&layout-url=https%3A%2F%2Fopen-source-webviz-ui.s3.amazonaws.com%2FdemoLayout.json)). See the help page in the top-right corner of the app for more customization for loading data into Webviz. You can probably even do this in an iframe if you really want to.

rgov, 2019-12-31:
Yes that's a good idea, but it is a concern that the public instance might not be compatible in the future. Suppose we are trying to load .rosbags from a decade ago, and everyone has been using ROS 2.0 for a long time, and you have come up with a new format for specifying layouts, etc. etc. and you don't want to be burdened with backwards compatibility issues.

Also we might want to customize it to only display a fixed set of views and remove the ability for the user to customize it, or add components for specific topics that our vehicles generate that are not useful to commit to the upstream project, etc.

So I would encourage you to think about releasing webviz-core as a regular NPM package and providing a way to integrate with other apps. The webviz.io site is wonderful for end-users but there's a lot of potential in making it possible to integrate.

Also, when we are taking this system out to sea, we would not be able to connect to webviz.io over our limited satellite uplink, so having a local copy is important. (But as mentioned I can build the docs site just fine so it's not a big issue.)

janpaul123, 2019-12-31:
Ah thanks for explaining!

> Yes that's a good idea, but it is a concern that the public instance might not be compatible in the future. Suppose we are trying to load .rosbags from a decade ago, and everyone has been using ROS [...]

We do intend to keep Webviz backwards compatible as much as possible. Internally we have no intention of switching to ROS 2.0 since we've heavily forked ROS 1.0, so hopefully we won't have too many issues there. And if there is a compatibility issue you can always check out an older version from our repo.

> Also we might want to customize it to only display a fixed set of views and remove the ability for the user to customize it, or add components for specific topics that our vehicles generate that are not useful to commit to the upstream project, etc.

Yeah that's a good point, but we try to as much as possible keep Webviz generic and instead express customizations like that in nodes. For example, you can use Python nodes to generate RViz markers, without having to customize RViz itself for your use case. For plots we have the generic 2D Plot panel where you can draw arbitrary plots. We also have a panel in Webviz called Node Playground where you can write nodes in the browser itself.

Of course if you need to heavily customize things, then that's always possible, but I'd generally discourage that since it makes it harder to resolve merge conflicts if you want to stay up to date with Webviz improvements that we publish.

> Also, when we are taking this system out to sea, we would not be able to connect to webviz.io over our limited satellite uplink, so having a local copy is important. (But as mentioned I can build the docs site just fine so it's not a big issue.)

Yeah for that I would recommend just checking out our gh-pages branch and statically hosting it. Or building the docs site like you said. 👍

janpaul123, 2020-01-01:
That said, I'll also think more about releasing it as an npm package, even with a limited API, just to make it easier to integrate. It shouldn't be too difficult.

janpaul123, 2020-03-26:
@rgov did you end up getting this to work?

rgov, 2020-03-26:
I think so, yes. I have a custom config module like this:

```javascript
var config = require("/webviz/webpack.config")
config.output.publicPath = "/static/webviz/"
module.exports = config
```

And I use a Docker container to run

```
npm run bootstrap

# Based on the "build" script defined in package.json, but we override the
# Webpack config to make some customizations
npx lerna run build
npx webpack --env.NODE_ENV=development --config /webpack.config.customized.js
```

It's not ideal but works alright.

ubicray, 2020-03-26:
@rgov I'm trying the same but it's not working for me
overall webpack builds it but doesn't serve it. and I don't get to see anything in the static folder afterwards either