cruise-automation/webviz #312: Live viewer bogs down after time and data

defied, 2020-01-06:
Currently, while running webviz connected to a rosbag websocket, the data collected remains on screen from time of start and beings to slow the web display down as time increases.

It would be nice to have an option for live telemetry allowing the time scale to slide along a specified range of time, reducing the amount of data being kept.

janpaul123, 2020-01-06:
That's a great idea! Which panels in particular do you have this issue with?

defied, 2020-01-07:
It's most easily recognized with the plotters, as they maintain the full history. I haven't tried the others, including raw data.

janpaul123, 2020-01-07:
Cool, yeah, makes sense. Most panels keep a fairly limited set of data, but the Plot panel currently always keeps an indefinite set. I believe that the rosout panel also keeps a large amount of data, but not unlimited. The 3d panel keeps an indefinite set if you just keep adding markers and never delete them, but that's in your control to fix. 

So perhaps if we just add a maximum number of datapoints to the Plot panel, that might be enough?

defied, 2020-01-07:
And possibly make it switchable? It's nice in small rosbags to see the whole trail of data, but when you get larger bags, it would bog down just like live data.

defied, 2020-01-07:
Although, I guess if you let us set max datapoints, we could do that ourselves.

orshefi, 2020-07-26:
Is there any progress with a solution for this issue? limit X axis entries?

troygibb, 2020-07-28:
Check out the plot panel settings:

[image]

There's an `X range` field for at least limiting the # of x axis entries from a visual standpoint.

As far as performance is concerned, we still don't have a way of limiting the number of points in the Plot panel. We've been shipping a set of performance improvements lately though, I'd try the latest release of Webviz if you haven't already! Let us know if you're still running into issues.