cruise-automation/webviz #314: Latching topics and the 3dPanel

cjds, 2020-01-07:
Hello,

FYI @dbking77. We're having a problem with the way the 3dPanel keeps data between scrubbing on the timeline and latched topics.

The problem comes down to the way the `seekPlayback` function is implemented [here](https://github.com/cruise-automation/webviz/blob/2e7db3aafffec39b541728668c97ce7d83eee007/packages/webviz-core/src/players/RandomAccessPlayer.js#L421). Because all the latched messages in the `rosbag.js` are found at time 0 and because when you use `seekPlayback` you go back in time by the SEEK_TIME when you scrub to a particular time, you lose all the latched messages from that time. For us this is particularly frustrating as our maps are latched. 

There are 2 ways we were thinking of to solve this issue:

1. We look for latched connections when we run `getMessages` https://github.com/cruise-automation/webviz/blob/2e7db3aafffec39b541728668c97ce7d83eee007/packages/webviz-core/src/players/RandomAccessPlayer.js#L437 and then just pick up the last message from them regardless of time
2. We implement something that goes through all the topics and gets the last message before this time for that topic.

janpaul123, 2020-01-07:
Ah very interesting. We don't use latched topics as such, but we do have similar issues with our own bags. I'm currently actually working on something that should enable (2), but it's a pretty major refactor of our data storage APIs, so it might take a while. But it's my current priority internally as well, so that works out nicely. 😄

cjds, 2020-01-08:
(2) would be the ideal solution for this I think so that works out. Let me know if you want any help on the smaller chunks. If you make them issues on this project I can help tackle some

bigredfrog, 2020-03-06:
+1 on a solution for this...

janpaul123, 2020-03-26:
@cjds @bigredfrog update: still working on this, and @jasonimercer also mentioned this to me, so there's definitely a lot of interest. I had to pause working on it for a little bit though as we've had quite a few bugs that I've had to look into instead.

Btw @cjds and @bigredfrog, I'm trying to get to know our open source community better, and would love to learn more about what you use Webviz for. Could you perhaps shoot me an email at jp.posma@getcruise.com?

cjds, 2020-03-26:
Sure will do

cjds, 2020-04-21:
Hey @janpaul123 any update on this?

janpaul123, 2020-04-22:
Not yet!

cjds, 2020-06-18:
Any updates on this?

janpaul123, 2020-06-18:
A workaround for you could be to republish the latched topics at a regular interval. Not a great solution but maybe that can help for now?

cjds, 2020-06-18:
That's actually interesting. So make a "node" that listens to those topics?

janpaul123, 2020-06-18:
Btw we also have a Slack workspace now if you want to talk to us more easily:  https://github.com/cruise-automation/webviz/issues/461

cjds, 2021-02-24:
Was trying to mess with this today and this is where I got till

In the below example, I was trying to use the global variable and a local variable in the node to try to republish the message. 

The issues I ran into were:

1. The global variables do not seem editable from inside a node. Once changed they would get reset on the next loop
2. Local variables to the function seem to be reset when the scrubber on the bottom is moved.

As such I think I'm either doing the repub wrong or it isn't possible with this feature

```
import { Input, Messages } from "ros";

type Output = {};
type GlobalVariables = {
  id: number;
  localization: Messages.nav_msgs__OccupancyGrid;
};

let x: Messages.nav_msgs__OccupancyGrid;
export const inputs = [
  "/atlas/localization",
  "/controller_costmap_updater_clock"
];
export const output = "/webviz_node/my_new_map";

// Populate 'Input' with a parameter to properly type your inputs, e.g. 'Input<"/your_input_topic">'
const publisher = (
  message:
    | Input<"/localization">
    | Input<"/controller_costmap_updater_clock">,
  globalVars: GlobalVariables
): Messages.nav_msgs__OccupancyGrid | undefined => {
  if (message.topic == "/atlas/localization") {
    log("Resetting message");
    x = message.message;
  }

  if (x !== undefined) {
    log(x.info.resolution);
    return x;
  }
  log("Variable is undefined");
};

export default publisher;

```

patrick-aoun, 2023-02-01:
Hey guys
Is there any update regarding the latching topics?

janpaul123, 2023-02-01:
You should check out https://github.com/foxglove/studio, they might have added it!