cruise-automation/webviz #318: missing dependencies on master branch

flynneva, 2020-01-10:
when following the instructions for installation, running "npm run build" results in an error because its missing two packages: popper.js and react-dnd-html5-backend.

running npm i popper.js react-dnd-html5-backend results in a clean build.

janpaul123, 2020-01-10:
Huh, weird, CI on master is green and only ran [a day ago](https://circleci.com/gh/cruise-automation/webviz/1898). 😕 But it does install things slightly differently. Could you perhaps make a PR with those dependencies fixed? In our root package.json we have everything in devDependencies btw.

flynneva, 2020-01-15:
sorry just getting back to this! yeah ill see if I can make a PR tonight

flynneva, 2020-03-16:
closing this because it looks like its been fixed...sorry for not doing this myself awhile back :smile:

closed 2020-03-16 by flynneva