cruise-automation/webviz #319: Detection of Accept-Ranges header is failing

rgov, 2020-01-10:
I have an unmodified, self-hosted copy of Webviz that is loading from a relative URL, so there should not be any CORS configuration to worry about. And indeed, I had this working just a moment ago, but something changed and now I am getting an error when loading my .bag:

    Error: Remote file does not support HTTP Range Requests.

Indeed, if I look in the Chrome inspector's Network tab, I see the request going to the correct URL, and getting a 200 OK response, but the inspector does *not* show the `Accept-Ranges: bytes` header.

However when I `curl` the same URL, it is definitely there:

```
< HTTP/1.1 200 OK
< Server: nginx/1.15.12
< Date: Fri, 10 Jan 2020 20:10:40 GMT
< Content-Type: application/octet-stream
< Content-Length: 1140248
< Last-Modified: Thu, 09 Jan 2020 20:48:16 GMT
< Connection: keep-alive
< ETag: "5e179190-116618"
< Accept-Ranges: bytes
< 
```

Is there some reason why Webviz isn't able to see this header?

The server is vanilla nginx with no crazy configuration. So it's not responding with different headers to different user agents or anything like that.

jtbandes, 2020-01-10:
The code which detects this support is fairly straightforward. Perhaps you can set a breakpoint here and see what the code receives in `response.headers`?

https://github.com/cruise-automation/webviz/blob/4c76bcba00b44b9d1fdf6e9108d5b7a6f58bed10/packages/webviz-core/src/dataProviders/BrowserHttpReader.js#L39

janpaul123, 2020-01-10:
Not sure, might be also different request headers that make the server respond differently?

rgov, 2020-01-10:
@janpaul123: I've fiddled with the curl request to make it exactly match what Chrome claims to be sending (e.g., Chrome does not send `Accept` but curl does), and I've verified they're both using HTTP/1.1. I can't repro it...

janpaul123, 2020-01-10:
Not sure then! Maybe add some logging to the webserver to see exactly what it receives and sends when Chrome makes the request?

rgov, 2020-01-10:
After looking through a proxy, it looks like when Chrome makes the request, it sends `If-None-Match` and `If-Modified-Since` headers, which cause the server to return `304 Not Modified`. For some reason Chrome developer tools show this as `200 OK`.

Anyway, because the response is actually `304 Not Modified` the server does not bother to send an `Accept-Ranges` header.

janpaul123, 2020-01-10:
Very interesting! We could try disabling the cache on the initial fetch request in https://github.com/cruise-automation/webviz/blob/4c76bcba00b44b9d1fdf6e9108d5b7a6f58bed10/packages/webviz-core/src/dataProviders/BrowserHttpReader.js#L39? See e.g. https://hacks.mozilla.org/2016/03/referrer-and-cache-control-apis-for-fetch/

rgov, 2020-01-11:
Could you assume it supports Accept-Ranges and request 1 byte, or `sizeof(struct rosbag_header)`, and error if the length doesn't match?

janpaul123, 2020-01-14:
Yes, that seems like a great idea too. Feel free to try that. If you can find something that fixes your case, I'd be happy to merge it (provided that it also keeps working with S3 and GCS which is what we've tested this with so far)!

rgov, 2020-01-17:
Actually I think that's not going to work because if the server doesn't support `Accept-Ranges` it will ignore the header and then you're going to get the entire file back which is a crappy way to test. I think the solution is to just disable caching for this header check.

janpaul123, 2020-01-17:
We already do that though: we request the entire file and then just immediately abort the request. This way we can get the file size without having to make a HEAD request (which would be yet another complication in setting up webservers properly). You can see us doing that here: https://github.com/cruise-automation/webviz/blob/4c76bcba00b44b9d1fdf6e9108d5b7a6f58bed10/packages/webviz-core/src/dataProviders/BrowserHttpReader.js#L32 So it wouldn't be too terrible to make another such request with only one byte (can even make that request in parallel to keep things fast).