cruise-automation/webviz #323: Publishing to ROS from Webviz

johuber, 2020-01-13:
Is there currently a way to publish anything from Webviz to ROS? I am referring to something equivalent to the "Tools" in RVIZ, where you can interact with a point or set a goal pose.

In case you don't have that yet, do you expect implementation of it to be straightforward or do you anticipate any major issues due to your architecture?

janpaul123, 2020-01-14:
Currently we don't have support for this in the open source version. The first step would be to implement these methods, which should be pretty straightforward: https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/players/RosbridgePlayer/index.js#L281-L282

Once that's done we can make our "Publish" panel open source, which we currently only have in our internal version. It allows publishing JSON strings as ROS messages.

Then after that we'd have to implement integration in the 3d panel for the tools that you're referring to. That's probably the hardest part since the 3d panel is quite complicated and heavily in flux at the moment. But these rviz-like tools should hopefully not conflict too much with other work that we're doing on it, so it might not be too bad. 😄 

Let me know if you want go give any of this a shot, so I can see if I can help. For example, I could put a branch up with the Publish panel if you want to try working on the first bit.

johuber, 2020-01-14:
That's more or less the answer that I expected, thank you.

Unfortunately, I currently don't have the resources to implement this. I still think it could be an interesting feature for many users who want to use Webviz as a replacement for RVIZ; maybe someone else in the community is interested in doing it..

janpaul123, 2020-04-09:
A first version of this (the "Publish" panel) is in the latest release! #396 Thanks for the request!

closed 2020-04-09 by janpaul123