cruise-automation/webviz #324: [feature request] record rosbag file of specific topics and export to local system

flynneva, 2020-01-13:
I'm not sure if this exists yet but it would be awesome to be able to export to a rosbag from webviz.
Selecting specific topics would be cool as well.

janpaul123, 2020-01-14:
Interesting, what is the use case that you have in mind? Are you thinking of recording data over a websocket and then storing it in a bag, as an alternative to `rosbag record`? Or would you want to slice existing bags, as an `rosbag filter`?

flynneva, 2020-01-15:
the real benefit I see for webviz in general is it is a quick-and-easy way for non-ROS user's to get their hands on data. for instance if webviz could record data then I could be messing with a sensor in a lab, see something cool as far as performance/testing/etc. and then want to share it with other people. if webviz could do rosbag record then I could shoot someone a URL with my websockets address + port and then they could immediately see what I was seeing. same goes for ros parameters too...if we could add that functionality then you could set up a room full of different sensors, all connected to a network and then have algo/SW people remote "login" and collect data for the most up-to-date firmware.

I would lean more towards a `rosbag record` webviz panel where you could select specific topics to record and the location to save it to. you could even add an option to save it to some cloud datastore for others to easily see...instead of saving it locally then having to upload the local file to the cloud datastore once its done.

jtbandes, 2020-01-15:
I'm not sure how this part is related to recording:

> if webviz could do rosbag record then I could shoot someone a URL with my websockets address + port and then they could immediately see what I was seeing

can't you do that already by having them use webviz with the same configuration, pointed at the same server you're currently looking at?

flynneva, 2020-01-15:
yeah sorry I forgot to add "and they could take a recording from there desk"....

janpaul123, 2020-01-16:
Cool, that makes a lot of sense, I like that vision. One blocker for this is using the raw ROS messages over the websocket, which we can work on once this gets released: https://github.com/RobotWebTools/rosbridge_suite/pull/452

janpaul123, 2020-04-09:
Alright, cbor-raw has landed in webviz with #396, so this should be possible to do now, though still not super straightforward since panels don't yet have access to the raw binary data. We'll have to do that for some other optimizations soon though.