cruise-automation/webviz #337: Support for multi touch for mobiles and tablets

artvichi, 2020-01-31:
Hello! What you do is awesome, guys!
We are happy to use your library in our company! 
One thing would be awesome to have - support for multitouch and mobile devices (+ tablets) 
Do you have something like that in webviz?
If not, maybe you can add the feature - it would be cool! 😄 
Many of our customers would like to interact with system with smaller devices...

jtbandes, 2020-01-31:
Many features already work on mobile devices. However, some core features like the 3D panel aren't optimized for it. We would be happy to review PRs if you want to contribute improvements in this area!

A good starting point would be the [regl-worldview package](https://github.com/cruise-automation/webviz/tree/20fc7c395b049ab862e39d07ce329db94b5f5e6f/packages/regl-worldview), particularly the CameraListener:
https://github.com/cruise-automation/webviz/blob/20fc7c395b049ab862e39d07ce329db94b5f5e6f/packages/regl-worldview/src/camera/CameraListener.js#L44-L45

janpaul123, 2020-01-31:
Also note that officially we only support Chrome, so it might be best to use the Chrome mobile browser.

jtbandes, 2020-01-31:
> it might be best to use the Chrome mobile browser

I'm not sure this would make much of a difference on iOS, where (last I heard) all browser apps are required to use the system-provided webview. It shouldn't hurt too much to make it work properly in Safari though :)

janpaul123, 2020-01-31:
Oof, I'm surprised that they still have that restriction (antitrust legislation and all…). It might not, but we definitely use some Chrome-specific features here and there, and Mobile Safari/Webkit is notorious for its bugs. 😭 Anyway, always happy to take PRs that fix things in other browsers, we just won't guarantee that we can officially support those browsers.

conrad, 2020-10-07:
On Android Chrome version 84.0.4147.136, the page is erroring out in `WorldviewContext.initialize`:
`Error: (regl) "oes_texture_float" extension is not supported by the current WebGL context, try upgrading your system or a different browser`. No more updates on Chrome available, and this also happens on the dev build.

The page loads successfully on Firefox on Android, but you can't zoom, rotate, or translate.

baycxh, 2020-12-22:
@conrad I got the same problem as you when I open webviz on android phone. I have tried several phones, but it didn't work. Have you got the solution? Thanks!
The Error is : (regl) "oes_texture_float".

conrad, 2020-12-22:
No solution here. I created some overlay buttons that change the camera position--not ideal. We have someone building out a redesign that doesn't include point clouds or as many polygons using ThreeJS.

JohanVer, 2021-05-05:
I got the same Error: `Error: (regl) "oes_texture_float" extension is not supported by the current WebGL context, try upgrading your system or a different browser`. 
Is there some solution for it yet? Using a mobile tablet seems to be one of the major usecases (at least for me), so I wonder why this thread is still open. 
As @janpaul123 pointed out there are a few chrome-specific functions used, but at the same time it seems that this is the only browser where webviz isn't starting.