cruise-automation/webviz #338: Time series plotting support

acadien, 2020-02-04:
Webviz is useful for quickly plotting of timeseries that are ros compatible. I frequently use it for debugging raw sensor output; for example here is the output from a magnetometer;
[image]

The problem is that points accumulate and eventually the entire browser begins to lag under the weight of rendering all of these points. Any solution that allows me to limit the number of points would be handy, some ideas;
* set the X-axis limit so that as new points come in, old points drop off the back 
* an option to wipe the plot from time to time
* set the maximum number of points rendered and simply drop old ones

janpaul123, 2020-02-04:
Yeah, this issue was also noted in #312. I think we're gonna go with an x-axis limit (the third option that you mentioned above doesn't work great if topics are publishing at different rates).

acadien, 2020-02-04:
Excellent thank you @janpaul123

acadien, 2020-02-04:
duplicate - https://github.com/cruise-automation/webviz/issues/312

closed 2020-02-04 by acadien