cruise-automation/webviz #341: Fail to bootstrap using node v12

IanBoyanZhang, 2020-02-08:
Hello Webviz team,

Thanks for releasing such an awesome tool! When I try setting project up with node v12.0.0 installed from nvm, I got below v8 error(s). 

Switched to v11.15.0, `npm run bootstrap` successfully completed bootstrap process. 

Looks like the error is related to 

[Issue 849 nan support for node 12](https://github.com/nodejs/nan/issues/849)

```
/Users/user/.node-gyp/12.0.0/include/node/v8.h:3416:3: note: 'Get' has been explicitly marked
      deprecated here
  V8_DEPRECATE_SOON("Use maybe version", Local<Value> Get(uint32_t index));
  ^
/Users/user/.node-gyp/12.0.0/include/node/v8config.h:322:29: note: expanded from macro
      'V8_DEPRECATE_SOON'
  declarator __attribute__((deprecated(message)))
                            ^
In file included from ../src/binding.cpp:1:
In file included from ../../nan/nan.h:2698:
_../../nan/nan_object_wrap.h:24:25: error: no member named 'IsNearDeath' in
      'Nan::Persistent<v8::Object, v8::NonCopyablePersistentTraits<v8::Object> >'
    assert(persistent().IsNearDeath());_
           ~~~~~~~~~~~~ ^
/usr/include/assert.h:93:25: note: expanded from macro 'assert'
    (__builtin_expect(!(e), 0) ? __assert_rtn(__func__, __FILE__, __LINE__, #e) : (void)0)
                        ^
In file included from ../src/binding.cpp:1:
In file included from ../../nan/nan.h:2698:
../../nan/nan_object_wrap.h:67:18: warning: 'MarkIndependent' is deprecated: Weak objects are always
      considered independent. Use TracedGlobal when trying to use EmbedderHeapTracer. Use a strong
      handle when trying to keep an object alive. [-Wdeprecated-declarations]
    persistent().MarkIndependent();
                 ^
/Users/user/.node-gyp/12.0.0/include/node/v8.h:563:3: note: 'MarkIndependent' has been
      explicitly marked deprecated here
  V8_DEPRECATED(
  ^
/Users/user/.node-gyp/12.0.0/include/node/v8config.h:307:29: note: expanded from macro
      'V8_DEPRECATED'
  declarator __attribute__((deprecated(message)))
                            ^
In file included from ../src/binding.cpp:1:
In file included from ../../nan/nan.h:2698:
../../nan/nan_object_wrap.h:124:26: error: no member named 'IsNearDeath' in
      'Nan::Persistent<v8::Object, v8::NonCopyablePersistentTraits<v8::Object> >'
    assert(wrap->handle_.IsNearDeath());
           ~~~~~~~~~~~~~ ^
/usr/include/assert.h:93:25: note: expanded from macro 'assert'
    (__builtin_expect(!(e), 0) ? __assert_rtn(__func__, __FILE__, __LINE__, #e) : (void)0)
                        ^
In file included from ../src/binding.cpp:1:
../../nan/nan.h:2378:9: warning: 'Set' is deprecated: Use maybe version [-Wdeprecated-declarations]
  recv->Set(name, GetFunction(tpl).ToLocalChecked());
        ^
../../nan/nan.h:2394:8: note: in instantiation of function template specialization
      'Nan::imp::SetMethodAux<v8::Local<v8::Object> >' requested here
  imp::SetMethodAux(recv, fn_name, t, static_cast<T*>(0));
       ^
../src/binding.cpp:351:8: note: in instantiation of function template specialization
      'Nan::SetMethod<v8::Object, Local>' requested here
  Nan::SetMethod(target, "render", render);
       ^
/Users/user/.node-gyp/12.0.0/include/node/v8.h:3358:3: note: 'Set' has been explicitly marked
      deprecated here
  V8_DEPRECATE_SOON("Use maybe version",
  ^
/Users/user/.node-gyp/12.0.0/include/node/v8config.h:322:29: note: expanded from macro
      'V8_DEPRECATE_SOON'
  declarator __attribute__((deprecated(message)))
                            ^
11 warnings and 3 errors generated.
make: *** [Release/obj.target/binding/src/binding.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/user/workspace/webviz/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:196:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:256:12)
gyp ERR! System Darwin 18.7.0
gyp ERR! command "/Users/user/.nvm/versions/node/v12.0.0/bin/node" "/Users/user/workspace/webviz/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "--verbose" "--libsass_ext=" "--libsass_cflags=" "--libsass_ldflags=" "--libsass_library="
gyp ERR! cwd /Users/user/workspace/webviz/node_modules/node-sass
gyp ERR! node -v v12.0.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
Build failed with error code: 1
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 (node_modules/fsevents):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 install: `node install`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-sass@4.10.0 postinstall: `node scripts/build.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-sass@4.10.0 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/user/.npm/_logs/2020-02-08T21_37_47_020Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! root@ bootstrap: `npm install && lerna bootstrap --hoist "{react,react-dom}"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the root@ bootstrap script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
```

janpaul123, 2020-02-10:
Yeah we should update our dependencies, so we get the fix from node-sass.

Sollimann, 2020-03-12:
I had the same issue. Solved by uninstalling node v13 and the install node v11 instead. Think it might work with node v10 as well.

janpaul123, 2020-03-26:
@IanBoyanZhang did you get this to work? Btw @IanBoyanZhang and @Sollimann, I'm trying to get to know our open source community better, and would love to learn more about what you use Webviz for. Could you perhaps shoot me an email at jp.posma@getcruise.com?

Sollimann, 2020-03-26:
@janpaul123 yea, got it to work. however, seems to be an issue in regards to rosbridge when subscribing to nav_msgs/Odometry /tf and joint_states. I get this error message: Exception calling subscribe callback: 'float' object has no attribute '__slots__' . I added a issue on it 28 days ago. As to your second question: I'm working in a Norwegian software company called Cognite that provides software solutions for the Oil & Gas industry. In recent time we have started a heavy investment in industrial robotics solutions. We wanted a web-based visualization tool for some of our robots running ROS middleware, and came across webviz.

janpaul123, 2020-04-09:
Cool, that makes a lot of sense, thanks for sharing! As for the other issue, that should be fixed now (see the other thread). I'll leave this ticket here open in case other people try to use node v12 (we'll have to make it compatible at some point!).

cjds, 2020-04-11:
Did confirm does work with Node v10.20.0 as well

jaguardo, 2020-12-09:
I'm kind of new to this, so sorry if this is not helpful! I got the same error and tried 
```
sudo npm install
```

and got similar errors, 

I then **deleted** package-lock.json and the node_modules and then bootstrap ran without errors (as well as the build process).  I'm having difficulties with getting the `npm run docs-deploy` to work however, but I'll post that in another topic.

vidaaudrey, 2020-12-10:
@jaguardo what's your node version? We support `>=10.22.0 <12`.

jaguardo, 2020-12-10:
copy that... I just installed the latest, so v15? I'll have to try downgrading, just not sure what else that will break.  I just thought it was interesting that I could get around the errors by deleting/re-install... obviously not a good long term solution, however it works... up until I got to ```npm run docs-deply``` but I think I might appreciate why that isn't working (stand alone machine with separate git repository)... thanks!