cruise-automation/webviz #346: Exception calling subscribe callback: 'float' object has no attribute '__slots__'

ubicray, 2020-02-27:
Why do I get this on the ROS side bridge when I try to plot?
What does that mean? Any idea?

jtbandes, 2020-02-27:
Can you provide some more details? Which version of ROS, what are you advertising on the topic, publisher side code, etc.

ubicray, 2020-02-27:
Ros Kinetic
Happens with a lot of kinds of topics but not all
For example if I try to plot latitude values from sensor_msgs/NavSatFix Message

janpaul123, 2020-02-27:
Ah I think this should get fixed once we implement the webviz side of this PR: https://github.com/RobotWebTools/rosbridge_suite/pull/452 We'll have to make some time for that soon, so stay tuned!

Sollimann, 2020-03-12:
I have the same issue when subscribing to /odometry /tf /joint_states messages. How's the fix doing? Btw, great tool you have here!

meitiever, 2020-03-16:
I have the same issue even if I subscribe to image topic.

janpaul123, 2020-04-09:
This should be fixed now since #396. You'll have to use the latest version of the rosbridge now. Let me know if you still have any issues!

closed 2020-04-09 by janpaul123