cruise-automation/webviz #347: no websocket connection is made for rosbridge

MaxwellGumley, 2020-02-28:
Summary:
The webvis app does not connect to rosbridge.  The help file indicates this should be automatic:
"By default we try connect to a rosbridge_server using a WebSocket on ws://localhost:9090."

Details:
I set up a rosbridge_server using the tutorial here:
http://wiki.ros.org/melodic/Installation/Ubuntu

I started the webviz on the same host and webviz demo works fine, but if I try to connect to the websocket of the rosbridge, I get no indication that a connection has been attempted.

Using a simple python script, I was able to confirm that the rosbridge is listening on ws://localhost:9090.
```
[INFO] [1582784076.170776]: Rosbridge WebSocket server started on port 9090
[INFO] [1582861849.478710]: Client connected.  1 clients total.
```

However, when I launch the webviz client, there is no indication in the rosbridge stdout that a websocket has connected.

Is there a way to view a log file to determine why it didn't connect?

Here is a screenshot of webviz, showing that it thinks it's using the websocket:
[image]

MaxwellGumley, 2020-02-28:
I found the issue.  My webvis server and rosbridge server are both on a remote machine.  I was using ssh forwarding so I can view the webpage on my local machine.  Ssh forwarding of localhost does not work for the rosbridge for some reason.  I'm still using ssh forwarding for webvis, but not for rosbridge.  I changed my url from this:
http://localhost:8020/app/?rosbridge-websocket-url=ws://localhost:9090
to this:
http://localhost:8020/app/?rosbridge-websocket-url=ws://192.168.10.35:9090
Now it works.

closed 2020-02-28 by MaxwellGumley

janpaul123, 2020-02-28:
Yay glad you figured it out!! 🎉