cruise-automation/webviz #348: ros2 bridge support

MaxwellGumley, 2020-03-04:
I have data in the ros2 format that I'd like to view with webviz.  I was able to connect webviz to my ros1 bridge (using ros1 data), but when I try to connect to the ros2 bridge (which has ros2 data), I get an error on the bridge server side:
```
$ ros2 run rosbridge_server rosbridge_websocket.py
registered capabilities (classes):
 - <class 'rosbridge_library.capabilities.call_service.CallService'>
 - <class 'rosbridge_library.capabilities.advertise.Advertise'>
 - <class 'rosbridge_library.capabilities.publish.Publish'>
 - <class 'rosbridge_library.capabilities.subscribe.Subscribe'>
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - <class 'rosbridge_library.capabilities.advertise_service.AdvertiseService'>
 - <class 'rosbridge_library.capabilities.service_response.ServiceResponse'>
 - <class 'rosbridge_library.capabilities.unadvertise_service.UnadvertiseService'>
[INFO] [rosbridge_websocket]: Rosbridge WebSocket server started on port 9090
[INFO] [rosbridge_websocket]: Client connected. 1 clients total.
[INFO] [rosbridge_websocket]: Client disconnected. 0 clients total.
[INFO] [rosbridge_websocket]: Client connected. 1 clients total.
[ERROR] [rosbridge_websocket]: [Client 1] [id: call_service:/rosapi/topics:1] call_service InvalidServiceException: Service /rosapi/topics
 does not exist
```
and an error on the web client side:
```
Error in fetching topics and datatypes
Service /rosapi/topics does not exist
```
I'm using the ubuntu package `ros-dashing-rosbridge-server`.  Please advise.

janpaul123, 2020-03-04:
We currently don't support ROS2. And I don't know too much about it, but happy to chat with folks on how we could support it.

janpaul123, 2020-03-04:
I guess I shouldn't close this since it is a good feature request that others might be interested in too!

flynneva, 2020-03-06:
so you can get this working by using the [ros1_bridge for ROS2](https://github.com/ros2/ros1_bridge) and then the normal ROS 1 bridge package.  its a little cloogy but it works :+1:

MaxwellGumley, 2020-03-07:
> 
> 
> so you can get this working by using the [ros1_bridge for ROS2](https://github.com/ros2/ros1_bridge) and then the normal ROS 1 bridge package. its a little cloogy but it works +1

Can you elaborate on this? I'm new to ROS.  My web client is using the ros1 websocket protocol.  For example, I need to query the websocket server for `/rosapi/topics` and `/rosapi/message_details`.

flynneva, 2020-03-08:
so the ros2 to ros1 bridge is a little funky to get working properly. 

you need to have sourced both the ros1 and ros2 setup.bash files you want to use (for me I used ros melodic and ros2 dashing). 

let me try and outline it in a list...ill assume you have ros melodic and ros2 dashing installed in your `/opt/ros` directory.

1.  Terminal 1 - source your ros 2 install if you havent already in your .bashrc and run the example talker node:
```bash
source /opt/ros/dashing/setup.bash
ros2 run demo_nodes_cpp talker
```

You should see it printing a counter going up with each message published.

2.  Terminal 2 - source your ros 1 install and start up roscore:
```bash
source /opt/ros/melodic/setup.bash
roscore
``` 
3.  Terminal 3 - source your ros 1 install and [start up ros1 bridge](http://wiki.ros.org/rosbridge_suite/Tutorials/RunningRosbridge):
```bash
source /opt/ros/melodic/setup.bash
roslaunch rosbridge_server rosbridge_websocket.launch
```

This will create a websocket for data to be passed to webviz through.

4.  Terminal 4 - source **both ros1 and ros2 installs** and start up the ros1_bridge following [these instructions from the ros1_bridge repo](https://github.com/ros2/ros1_bridge):
``` bash
 source /opt/ros/dashing/setup.bash
source /opt/ros/melodic/setup.bash
ros2 run ros1_bridge dynamic_bridge
```

And now open up webviz like normal and tell it to listen to the websocket.....anddd hopefully you see the ROS2 demo node publishing its messages!

Hope this helps!

flynneva, 2020-03-08:
I think for a longer term vision of webviz it would make more sense to follow the progress of [rclnodejs](https://github.com/RobotWebTools/rclnodejs) and once it has all or most of the features that webviz needs just have a forked version of webviz that uses that library instead.

not sure of the effort for this but my guess is it would be pretty high.....

janpaul123, 2020-03-09:
Unfortunately internally in Webviz we're pretty tied to the ros1 binary message format, and we're going to lean on it more for performance improvements soon. So we'll have to figure that part out.

janpaul123, 2020-03-26:
Btw @MaxwellGumley @flynneva, I'm trying to get to know our open source community better, and would love to learn more about what you use Webviz for. Could you perhaps shoot me an email at jp.posma@getcruise.com?

alexApplegreen, 2021-02-23:
@janpaul123 could you give an update on this? Much appreciated!

janpaul123, 2021-02-26:
I'm not working on Webviz any more, sorry!

maxpolzin, 2022-06-27:
Thank you for the previous responses.

The described workarounds ROS2->ROS1->rosbridge->webviz work well, thank you.

Does anyone know if webviz works with the https://github.com/RobotWebTools/ros2-web-bridge?
Are you accepting Pull-Requests for Webviz? I am considering to enable communication with the rosbridge_suite of ROS2.
If you are no longer working on webviz, is it currently not maintained?

janpaul123, 2022-06-27:
I'd suggest checking out https://foxglove.dev/

maxpolzin, 2022-07-05:
Thx for your fast response @janpaul123. Somehow, I missed the notification.

I am frequently using https://foxglove.dev/ and it is an amazing successor of webviz. 
However, for the current project I need to be able to import a layout.json from a website into my visualiser. 
While webviz supports this, afaik, in foxglove this can only be done by logging in and layouts are only shareable organization-wide.  

Is there a setting in Foxglove's URL to load a specific layout.json which is hosted on a thirdparty host or to load a "publicly" shared layout via layout-id?

jtbandes, 2022-07-05:
For questions about Foxglove Studio, please join our GitHub or Slack community: https://foxglove.dev/community  You can add your input on this ticket: https://github.com/foxglove/studio/issues/2792