cruise-automation/webviz #350: npm run build error

heshiwen1989, 2020-03-08:
Hi, webRiv Teams: 
 when i follow git readme to use this project, i get error, e.g. 
ERROR in ./packages/webviz-core/src/components/Modal.js
Module not found: Error: Can't resolve 'react-key-listener' in '/home/xxx/Downloads/webviz-master/packages/webviz-core/src/components'
 @ ./packages/webviz-core/src/components/Modal.js 12:47-76
 @ ./packages/webviz-core/src/components/ShareJsonModal.js
 @ ./packages/webviz-core/src/components/PanelToolbar/index.js
 @ ./packages/webviz-core/src/panels/diagnostics/DiagnosticSummary.js
 @ ./packages/webviz-core/src/loadWebviz.js
 @ ./packages/webviz-core/src/index.js

chengtz, 2020-03-08:
老师，他这个不能直接npm install来安装所有依赖，他在readme中说了Please see the individual package READMEs for details on how to install and use them.  请查看这些支持包自己的readme来安装部分依赖和使用

jtbandes, 2020-03-08:
Hello, did you try running `npm run bootstrap` first? Can you please share the full console log including the commands you ran?

heshiwen1989, 2020-03-08:
> Hello, did you try running `npm run bootstrap` first? Can you please share the full console log including the commands you ran?

My Step is: 
Step1: nmp run bootstrap, but install node-sass get some error, and then i install node-sass independently by command: nmp install node-sass --save, finally run nmp run booststrap again no any error; 
Step2: nmp run build , and get error: 
[image]

heshiwen1989, 2020-03-08:
> Hello, did you try running `npm run bootstrap` first? Can you please share the full console log including the commands you ran?

My Step is:
Step1: nmp run bootstrap, but install node-sass get some error, and then i install node-sass independently by command: nmp install node-sass --save, finally run nmp run booststrap again no any error;
Step2: nmp run build , and get error:
hsw@hsw-ThinkPad-E460:~/Downloads/webviz-master$ npm run build

> root@ build /home/hsw/Downloads/webviz-master
> lerna run build && webpack

lerna notice cli v3.6.0
lerna info Executing command in 5 packages: "npm run build"
lerna info run Ran npm script 'build' in '@cruise-automation/rpc' in 2.4s:

> @cruise-automation/rpc@0.0.2 build /home/hsw/Downloads/webviz-master/packages/@cruise-automation/rpc
> NODE_ENV=production npm run build-dev && flow-copy-source -v src lib --ignore '*.test.*'

> @cruise-automation/rpc@0.0.2 build-dev /home/hsw/Downloads/webviz-master/packages/@cruise-automation/rpc
> BABEL_ENV=$NODE_ENV babel src/index.js --out-dir lib --copy-files --config-file ../../../babel.config.js

Successfully compiled 1 file with Babel.
src/index.js -> lib/index.js.flow
lerna info run Ran npm script 'build' in '@cruise-automation/button' in 2.7s:

> @cruise-automation/button@0.0.7 build /home/hsw/Downloads/webviz-master/packages/@cruise-automation/button
> NODE_ENV=production npm run build-dev && flow-copy-source -v src lib --ignore '*.test.*'

> @cruise-automation/button@0.0.7 build-dev /home/hsw/Downloads/webviz-master/packages/@cruise-automation/button
> BABEL_ENV=$NODE_ENV babel src/index.js --out-dir lib --copy-files --config-file ../../../babel.config.js

Successfully compiled 1 file with Babel.
src/index.js -> lib/index.js.flow
lerna info run Ran npm script 'build' in '@cruise-automation/hooks' in 3.2s:

> @cruise-automation/hooks@0.0.1 build /home/hsw/Downloads/webviz-master/packages/@cruise-automation/hooks
> rollup -c

lerna info run Ran npm script 'build' in '@cruise-automation/tooltip' in 2.2s:

> @cruise-automation/tooltip@0.0.7 build /home/hsw/Downloads/webviz-master/packages/@cruise-automation/tooltip
> NODE_ENV=production npm run build-dev && flow-copy-source -v src lib --ignore '*.test.*'

> @cruise-automation/tooltip@0.0.7 build-dev /home/hsw/Downloads/webviz-master/packages/@cruise-automation/tooltip
> BABEL_ENV=$NODE_ENV babel src/index.js --out-dir lib --copy-files --config-file ../../../babel.config.js

Successfully compiled 1 file with Babel.
src/index.js -> lib/index.js.flow
lerna info run Ran npm script 'build' in 'regl-worldview' in 19.9s:

> regl-worldview@0.4.0 build /home/hsw/Downloads/webviz-master/packages/regl-worldview
> rollup -c

lerna success run Ran npm script 'build' in 5 packages in 19.9s:
lerna success - regl-worldview
lerna success - @cruise-automation/button
lerna success - @cruise-automation/hooks
lerna success - @cruise-automation/rpc
lerna success - @cruise-automation/tooltip
Browserslist: caniuse-lite is outdated. Please run next command `npm update caniuse-lite browserslist`
Hash: a0def8f42bf2840e0668
Version: webpack 4.27.1
Time: 31653ms
Built at: 03/08/2020 10:55:49 AM
                                            Asset      Size                          Chunks                    Chunk Names
                                             0.js  25.1 KiB                               0  [emitted]         
                   09517e8c28a52c2dcbfd.worker.js    60 KiB                                  [emitted]         
                                             1.js   390 KiB                               1  [emitted]  [big]  
                                            10.js  17.8 KiB                              10  [emitted]         
                                            11.js  20.3 KiB                              11  [emitted]         
                                            12.js  6.95 KiB                              12  [emitted]         
[... 189 lines cut ...]
    [0] fs (ignored) 15 bytes {main} [built]
    [1] path (ignored) 15 bytes {main} [built]
    [2] os (ignored) 15 bytes {main} [built]
    [./node_modules/babel-loader/lib/index.js?cacheDirectory!./packages/webviz-core/src/players/UserNodePlayer/nodeTransformerWorker/index.js] ./node_modules/babel-loader/lib?cacheDirectory!./packages/webviz-core/src/players/UserNodePlayer/nodeTransformerWorker/index.js 855 bytes {main} [built]
    [./node_modules/webpack/buildin/global.js] (webpack)/buildin/global.js 472 bytes {main} [built]
    [./packages/webviz-core/src/players/UserNodePlayer/nodeTransformerWorker/transformer.js] 12.4 KiB {main} [built]
    [./packages/webviz-core/src/players/UserNodePlayer/nodeTransformerWorker/typescript/ast.js] 13.7 KiB {main} [built]
    [./packages/webviz-core/src/players/UserNodePlayer/nodeTransformerWorker/typescript/baseDatatypes.js] 5.79 KiB {main} [built]
    [./packages/webviz-core/src/players/UserNodePlayer/nodeTransformerWorker/typescript/lib.js] 259 KiB {main} [built]
    [./packages/webviz-core/src/players/UserNodePlayer/nodeTransformerWorker/typescript/ros.js] 3.65 KiB {main} [built]
    [./packages/webviz-core/src/players/UserNodePlayer/nodeTransformerWorker/utils.js] 2.7 KiB {main} [built]
    [./packages/webviz-core/src/players/UserNodePlayer/types.js] 1.42 KiB {main} [built]
    [./packages/webviz-core/src/util/Rpc.js] 5.19 KiB {main} [built]
    [./packages/webviz-core/src/util/globalConstants.js] 5.34 KiB {main} [built]
    [./packages/webviz-core/src/util/reportError.js] 2.69 KiB {main} [built]
        + 177 hidden modules
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! root@ build: `lerna run build && webpack`
npm ERR! Exit status 2
npm ERR! 
npm ERR! Failed at the root@ build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:

heshiwen1989, 2020-03-08:
> 老师，他这个不能直接npm install来安装所有依赖，他在readme中说了Please see the individual package READMEs for details on how to install and use them. 请查看这些支持包自己的readme来安装部分依赖和使用

我看他的readme不是说这个工程包含很多部分，并且，我看src中包含react-key-listener相关的东西，能否指导一下呀

heshiwen1989, 2020-03-08:
Hello, My Solution for this issue: 
Use npm install react-key-listener directly(by: npm install react-key-listener --save), and Then No Any Error, Thanks!

chengtz, 2020-03-08:
老师这个东西错误是真的多，我好好弄下，给他弄docker封装到国内容器里，到时候直接使用了

chengtz, 2020-03-08:
找到了 docker pull johannhaselberger/webviz   不过是六个月前的
何老师你npm安装好了没？ 安装好能留个联系方式吗

heshiwen1989, 2020-03-08:
> 找到了 docker pull johannhaselberger/webviz 不过是六个月前的
> 何老师你npm安装好了没？ 安装好能留个联系方式吗

我是直接git下载这个工程，然后按照 npm run bootstrap，如果报错，就通过npm install 出错模块 --save 单独安装出错的模块， 然后 npm run build, 我遇到的是找不到模块react-key-listener, 我就通过npm 直接安装这个模块，再次npm run build 就ok了

closed 2020-03-08 by heshiwen1989

chengtz, 2020-03-08:
老师。我刚才一分钟就安装好了，用的是docker，别人已经装好的依赖 

https://hub.docker.com/r/johannhaselberger/webviz

直接用docker映射8080就可以直接使用了，你可以保存一下，可以给个联系方式吗，我是南京金陵科技学院的学生，有ROS方面的问题可以请教请教

chengtz, 2020-03-08:
手动也装好了，给家里服务器连到路由器装了个翻墙软件

heshiwen1989, 2020-03-08:
> 手动也装好了，给家里服务器连到路由器装了个翻墙软件

你可能是网络不好也有可能，npm比较慢可能会有一些失败；你打算用这个做什么

chengtz, 2020-03-08:
毕业设计就是设计ROS小车，弄个可视化的界面再加上自己新增的一些传感器话题发布，到时候导师就很容易通过了

vidaaudrey, 2020-03-09:
@chengtz 除了`react-key-listener`, 你还发现了什么错误呢？ 我刚刚重新在本地安装了一遍没发现什么问题：`git clone...` => `npm run bootstrap` => `npm run build` => `npm run docs`. 

另外你能告诉我你的 npm, node 的版本吗？我的是 npm(6.13.4), node (v10.15.3). 你用的是什么操作系统呢？