cruise-automation/webviz #352: How to switch from chartjs for better performance

ubicray, 2020-03-09:
Great job with webviz, there's no comparable package that works just as well

I have an issue with the plotting tho, as plotting something at 100 hz for a minute can break the back of my i7 processor in a minute, slowing the performance down to 0.2 fps
I tried simplifying the plots with simplifyjs, but the issue wasn't solved with it

I wanna give plotly a chance (https://plot.ly/~cal8food/28/render-times-of-plotly-and-chartjs/#plot)
Has anybody tried to change the plotting library? How should one go about that without breaking everything else?

Best Regards

davidcrawford, 2020-03-09:
Hi Elmar,

We are going to land a big performance improvement to the plot panel soon, moving the chart rendering to a web worker, but this is an interesting comparison that I don't think we've considered.  One thing you can try as an experiment is adding a separate plotly panel to see if you can really get these gains.  Performance is a high priority for us, so we'd be happy to help incorporate it if you see promise in this direction.

Glad you're getting some use out of Webviz!

Best,
David

davidswinegar, 2020-03-09:
Hi @ubicray - as David stated, we are moving the chart calculation behind a web worker soon, so that should reduce some load for your charts. We will also be switching to use Chart.js 3.0 (https://github.com/chartjs/Chart.js/issues/6598) when that is released in the next few months which should give you a massive speedup in these cases.

If you are interested in making your own panel with better performance, switching to the Chart.js version on master or using uPlot (https://github.com/leeoniya/uPlot) would probably be your best bets for a really efficient plotting framework.

janpaul123, 2020-04-09:
Charts are now plottted from a WebWorker, since #396. Let us know if you have any more suggestions. It would be helpful if you can share a bag that is slow to render for you, so we can have a look at it. For now I'm closing this ticket!

closed 2020-04-09 by janpaul123