cruise-automation/webviz #353: Need Guidance

walstanb, 2020-03-10:
How to I use webviz 2D plot to plot a point path in a 2D plane

janpaul123, 2020-03-10:
There's some documentation and an example [here in the docs](https://github.com/cruise-automation/webviz/blob/e70c549edcdcbf9f06b90c43afe7915b8395c2b0/packages/webviz-core/src/panels/TwoDimensionalPlot/index.help.md). The part under "message" is what the message itself should look like.

cc @esthersweon @davidswinegar maybe you have more examples?

walstanb, 2020-03-10:
Thank you, could you help me out with examples. I just have two constant values from the whycon localization system and I want to plot them on a graph continuously to form a path

janpaul123, 2020-03-10:
Maybe first try to get this first example message to render, and from there you can add more lines/points/polygons to get the plot that you need.

walstanb, 2020-03-10:
how do I do that, do I need to make a .msg file as the example and publish it ?

walstanb, 2020-03-10:
and is it possible to get this the full thing working in 3D and make a path in 3 dimensional space ?

janpaul123, 2020-03-10:
> how do I do that, do I need to make a .msg file as the example and publish it ?

Exactly.

> and is it possible to get this the full thing working in 3D and make a path in 3 dimensional space ?

For that you should use the 3D panel and publish [RViz markers](http://wiki.ros.org/rviz/DisplayTypes/Marker).

closed 2020-10-24 by walstanb