cruise-automation/webviz #358: Access webviz from mobile devices

Reaper151, 2020-03-11:
I tried to access webviz from mobile devices in same network by using 
`webpack-dev-server --host <ip-of-my-pc-from-WIFI-properties> --port 8080` instead of `npm run docs` since I need IP of web to access but I encountered some problems
1.Layout in mobile and host PC not match
2.I can't find any topics in mobile while I found them in PC

Additional feature which I'd like to see : auto-scroll x axis (like rqt-plot) by configurable period of time to display

Reaper151, 2020-03-11:
UPDATED: I can access by forward ROS bridge server by this command
`<launch>
  <include file="$(find rosbridge_server)/launch/rosbridge_websocket.launch" > 
     <arg name="port" value="_your-port-number_"/>
     <arg name="address" value="_your-ip-address_"/>
  </include>
</launch>`
Credit: http://wiki.ros.org/rosbridge_suite/Tutorials/RunningRosbridge

and I can sync layout by export layout setting from host PC and save it as json then upload and use this URL format http://xxx.xxx.xx.xx:8080/app/?layout-url=https%3A%2F%2Fopen-source-webviz-ui.s3.amazonaws.com%2FdemoLayout.json

janpaul123, 2020-03-11:
Glad that you managed to figure it out!

> Additional feature which I'd like to see : auto-scroll x axis (like rqt-plot) by configurable period of time to display

Yeah, that's being tracked here: #312.

closed 2020-03-11 by janpaul123

Reaper151, 2020-03-11:
But I can't sync layout by export layout setting from host PC and save it as json then upload to google drive and use this URL format http://xxx.xxx.xx.xx:8080/app/?layout-url=https%3A%2F%2Fopen-source-webviz-ui.s3.amazonaws.com%2FdemoLayout.json.

janpaul123, 2020-03-11:
Google Drive probably doesn't set the right [CORS headers](https://github.com/cruise-automation/webviz/blob/e57b3a17f6caafd2d152696c220a6c24888f83e9/packages/webviz-core/src/util/corsError.js#L10). You'll need to use a server that sets the correct CORS headers. I'd recommend S3 or GCS.

janpaul123, 2020-03-11:
Or you can just manually copy/paste the layout.