cruise-automation/webviz #361: Current status of cbor compression usage?

maxlein, 2020-03-11:
Hi,

I just saw the closed PR on the rosbridge repo and a comment that some work has still to be done on the webviz side.  

Is this already working or not finished yet?  

Because I saw this, and it looks like its already used:  
https://github.com/cruise-automation/webviz/blob/e57b3a17f6caafd2d152696c220a6c24888f83e9/packages/webviz-core/src/players/RosbridgePlayer/index.js#L252

janpaul123, 2020-03-11:
Ah no, this uses the old cbor compression, we still need to switch to cbor-raw. I haven't gotten around to it yet!

janpaul123, 2020-03-26:
@maxlein I've written the code for this and am going through our internal review process. I'll keep you posted! Btw, I'm trying to get to know our open source community better, and would love to learn more about what you use Webviz for. Could you perhaps shoot me an email at jp.posma@getcruise.com?

janpaul123, 2020-04-09:
This is in the latest release! #396 Thanks for the report!

closed 2020-04-09 by janpaul123