cruise-automation/webviz #371: How to make ros service calls

ubicray, 2020-03-17:
So far what I have in mind:
Create a new panel type
Put buttons in it
Make buttons call services via rosbridge
Is there any documentation regarding that I can use?

janpaul123, 2020-03-17:
Yeah this is a great request, and very similar to #323. Like I said in the answer there, in our internal version we have similar functionality, so I should really just allocate some time soon to port it to the open source version…

ubicray, 2020-03-23:
> Yeah this is a great request, and very similar to #323. Like I said in the answer there, in our internal version we have similar functionality, so I should really just allocate some time soon to port it to the open source version…

Thanks for letting me now!
I'd rather need it ASAP, so started implementing it myself. Would you mind sharing a panel with service integration if possible?

janpaul123, 2020-03-25:
We currently don't have a panel with service integration, but tomorrow I'll push our internal "Publish" panel with which you can publish on arbitrary topics. If you follow the logic of how that one works it should be somewhat straightforward to modify it to support service calls as well. I'll update you here when I push that code.

janpaul123, 2020-03-25:
@ubicray FYI pushed it here: https://github.com/cruise-automation/webviz/pull/378

janpaul123, 2020-04-09:
The "Publish" panel is in the `master` branch now. It should be pretty straightforward to extend it to support service calls; let me know if you need any help with that!

rgov, 2020-12-08:
Has there been any progress here? Or, what would it take to extend this for service calls?

troygibb, 2020-12-09:
Hi @rgov! To my knowledge we haven't updated the Publish panel to make service calls. Have you tried tweaking it as JP suggests above?