cruise-automation/webviz #373: Slightly broken bags no longer work after #354

jasonimercer, 2020-03-19:
Before #354, bags with missing message definitions would play in webviz. At the command line, we get a warning but things generally work (rosbag play):

```
[ WARN] [/play_1584635435660808555] [1584635435.697884385]: Advertising on topic [/led_array_status] with an empty message definition.
```

After #354, webviz now fails to initialize with:

```
Error initializing player
Error: ApiCheckerDataProvider assertion failed: ApiCheckerDataProvider(BagDataProvider@3) error: Topic "/led_array_status"" not present in messageDefintionsByTopic
    at ApiCheckerDataProvider._error (webpack-internal:///./packages/webviz-core/src/dataProviders/ApiCheckerDataProvider.js:175:13)
    at ApiCheckerDataProvider.initialize (webpack-internal:///./packages/webviz-core/src/dataProviders/ApiCheckerDataProvider.js:90:14)
```

I expect there's a way we could repair our bags internally but we now rely on webviz for our tier 1 - 3 triage teams (sorry, I should have prefixed all this with gushing omg thank you so much for webviz). We have lots of data on our servers that's automatically converted to webviz-compatible formats and stream everything with the remote-url feature, it would be handy if we could continue using historic bags without reencoding everything.

I've created a minimum working example:

```
begin 664 robot_ph_cpe22_192_81590368_base_4_webviz.unit_test_topic.bag.bz2
M0EIH.3%!62936>ZI"YP``75______M#)H?^B>I*9@[_O_W@+&@(0.^`$2"D1
MHH*"D"@(0`).[,`-89)3R2:/TH`VH`/4``T`:&@`````!H`T-!^J'`````-`
M````&@T````#(`T``8J4#1H&30,AZ0,@!HP@:#1H::#0`!H8`"``52),(IY#
M0RFTIL:IDT]3(]1B:,`C!#1HR,1ZCTTT-";()ZFR:+@@",^VL:M2('T2MI(&
M3H(`>!-[#86(90("4&2@4-ABFB,:]H2RH'"S&A3-F-%GD*YZ*(QGQN"W-F_?
M))%PB2R422`X!9#"-<P"20&2/*+C/ORCJY7=-^&M!*:."EW$6A`$"!56@_9B
M@BM#X5#%IV2RQ3U](LXH)B#86KZPIY@D+>\0P$ZXAHTF260)`4'.$ESA4!&#
MX:XDVND#\Y!/K;63N^^GLR(VU=55=Y9T+Z3=[/`M,M$WY3?TXSK..[_UJ%QQ
MZ':-]^KBI*EU8M*VG)G+R;2>*!Y?-$O*0GAM.`USI-ED135,-^-77QNU_<S4
MU]77_EJ:T5ME$1FB_/*;9/;F<&KC8]%GLY)Y!7YTD:(TF4\YXUU4E"%DI2^S
MY'RN&L8!A_UJ+BA@4K;TZ:X&FS169/I4*6"FXXMUD,YG%XL8`O;3:4*[:6ZB
M(FJ:#B(OKSU*6(),6);@3B6931(J304KE!4RZHG-!80R#+4%LAC2+C#L-)D,
M(MS94&29ZXP+B637$KIREB^7:_H2+JQ9(B68F4(DM,2\9J+2]4*!9;4$4@X%
M]5'\*HIV!V0A8VZ";!AA))))))()*7>G.26TWFRDW5#>2-:$U$4MQW_:WF'G
>R;[<:J.'!F;'.\$':'!;@X/DG_%W)%.%"0[JD+G`
`
end
```

that triggers the issue we've come across. For now, we're OK running a version of webviz ahead of merge but it would be nice to be able to get back to HEAD and keep using + testing all your neat improvements.

Thanks again for this amazing contribution to the community.

janpaul123, 2020-03-19:
Oh interesting! Weird how a topic with no message definition can end up in the bag, because by definition that means that any messages on that topic can't get parsed. Anyway, I think we can fix this pretty easily on our end; I'll have a look soon!

Btw, thanks so much for the kind words! We don't have too much insight in who uses Webviz so it's really cool to hear that we're such an integral part of your workflow. I always love to learn more about our power users, so if you're down to have a chat some time could you please contact me at jp.posma@getcruise.com? :-)

janpaul123, 2020-04-09:
This is in the latest release! #396 Thanks for the report!

closed 2020-04-09 by janpaul123